bitdump/BLHeli #265: Littlebee Summer 30A Missing RECORD_MARK at filestart error. Operation Aborted

Darozas, 2017-10-21:
Hi all trying to flash my esc with test code without success keep getting this error msg

Hex File Read Error. Missing RECORD_MARK at filestart error. Operation Aborted

Using Blheli 32.3.0

How can i over come this problem???

Regards

4712, 2017-10-21:
Download the file using the right mouse button on "Raw" and choose something like "save target as..."

Darozas, 2017-10-21:
Thanks for reply worked one more quick question

I do own 30A esc but Blheli recognize them as 25A do i still flash to 30A version???

Regards

4712, 2017-10-21:
Forget about the 30A in the filename. It is just the name of the layout, doesn't stand for the actual current capacity.

Darozas, 2017-10-21:
Thanks for reply

So im ok to flash 30A version over 25A version?

Regards

Darozas, 2017-10-21:
Just checked and foud that i cant flash 30A version 

So been sold fake A rating or what going on here then???

Regards

closed 2017-10-22 by Darozas