bitdump/BLHeli #267: Dshot beeper 16.67RC

SlipstreamFPV, 2017-10-25:
When the dshot beeper command is sent BLHeli_S 16.67RC uses the startup beep volume not the beacon volume. This does not happen in 16.63 version. If you have the startup volume set low then the beeper is also extremely quiet defeating its purpose.

nmaggioni, 2017-10-26:
I have the same problem: beep strength controls both the startup tone (that I would like to keep relatively quiet in order to avoid constantly harassing other pilots when changing batteries) and beeper (that should be, to be useful, at a way higher value).

Is there any chance that these two strength settings can be decoupled?

@SlipstreamFPV I'd suggest changing the title to something more descriptive, like _"DShot beeper uses startup strength value in v16.67RC"_, or nobody will notice this issue.

fiveangle, 2017-11-23:
See https://github.com/bitdump/BLHeli/issues/234#issuecomment-330103003