bitdump/BLHeli #271: Typhoon32 (32.3 hex) Dshot1200, turtle mode works, beeper does not

jiffypoo, 2017-10-31:
Betaflight 3.2.1
Latest blheli32 software
Typhoon32 (32.3 hex) - Tried 1200, 600, and 300 in Betaflight 3.2.1 with same result

Turtle mode works, beeper does not.

4712, 2017-10-31:
@jiffypoo did to activate the Beeper switch in BF Configurator?
[image]

jiffypoo, 2017-10-31:
@4712 Yes the Beeper mode is activated in the BF Configurator and the Mode will activate yellow in BF 3.2.1, as well as the LED indicator blinking repeatedly to show the Beeper mode is active on the BetaflightF4 flight controller.

4712, 2017-10-31:
@jiffypoo I checked it here, and it is working fine. Using BF Release 3.2.1 with DShot600 and Typhoon32 32.3...

jiffypoo, 2017-10-31:
@4712 your suggested CLI updates fixed the issue!  Thanks!  Hope this can be integrated into the GUI in future updates.

closed 2017-10-31 by jiffypoo