bitdump/BLHeli #274: ESC Beacon not working without FC

habonlaci, 2017-11-05:
Is it on purpose that the ESC beacon funcionality is only active when an FC is connected? 
I think since it is a BLHeli feature and has nothing to do with the FC it would be very helpful if it would activate once the timer is up even if there is no FC present at all.

mikeller, 2017-11-05:
I think @habonlaci has a good point there. One rationale behind the timer based ESC beacon is to serve as a 'last resort' acoustical locator after a crash, so it would make sense for it to activate even if the connection to the flight controller has been severed or the flight controller / voltage regulator has been damaged in the crash.

habonlaci, 2017-11-05:
Exactly. Today I crashed in a tree and somehow, the connection between the FC and ESC (4in1) got ripped out. Took me like 1 hour to find my quad in the tall grass, and I was very lucky. I use dshot beeper what obviously was not working either.

sskaug, 2017-11-06:
It turns out that this can be overly complicated. When there is no input connected, the input pullup resistor causes code to go into bootloader mode. Where it will stay as long as there is no input signal. So from the way the system and code currently is designed, activating the beacon with no input signal will be quite complex.

mikeller, 2017-11-06:
That's not ideal indeed. Would it be an option to add the beacon timeout to the bootloader, and make it jump to the beacon routine after the timeout has elapsed?

sim-, 2018-07-03:
Hrm.. Why is there a pull-up on the input, anyway? That seems like it might be a bit dangerous for loose connections.

sskaug, 2018-07-04:
Pullup is chosen in order to work with the pullup of the one-wire bootloader interface. A pulldown would work against the (dongle or FC) pullup.

sim-, 2018-07-04:
Hello :) Isn't the one-wire bootloader supposed to be the thing doing the pull-up, though? Not sure if you're talking about the old turnigy linker thing, but for that I had to use a weak pull-down (1M) to still be able to detect its own weak pull-up while still having something to otherwise drain the pin from noise when not connected.

pranderish, 2018-07-05:
ROFLMFAO!!!!!!!!
@sskaug, dude, listen to @sim-. He actually knows what hes doing. Maybe if u ask nicely u can get @kidBrazil to help u make a real bootloader that actually works correctly.