bitdump/BLHeli #277: Use only Digital protocol option

evnngsky, 2017-11-09:
Sometimes pdb noise make signals like legacy PWM, So it make motors run without command and it make some injury.
And so many users select digital protocol for most ESC, So new option is needed, I guess.

If add option to initialize with only digital protocol, It can prevent serious injuries.

Thanks.

richard-scott, 2017-11-09:
If your motors move when disarmed and your not running Dshot then you need to lower your ESC disarm signal by ajusting the "Minimum command" in this screenshot.

[image]

evnngsky, 2017-11-09:
My opinion is not for moving after disarming.
Sometimes, some noise make ESC initialize with wrong protocol before FC send signal.
I want to prevent faulty initialized ESC make motor move.

Hrumque, 2018-01-27:
I join the above proposal.
There should be an option to select a protocol in the configurator:
- automatically (default, as it is now, automatically detected protocol during ESC start)
- only one specific protocol chosen (eg there would be no problem with false detection during the startup)