bitdump/BLHeli #281: 16.67-dshot `beeper_dshot_beacon_tone` executes at "beep" strength and not "beacon" strength

fiveangle, 2017-12-11:
As [reported here](https://github.com/bitdump/BLHeli/issues/234#issuecomment-321837596), `beeper_dshot_beacon_tone`still executes at same strength as startup tones, and not at strength of "beacon" as set in BLHeli Suite/Configurator.

@sskaug [said this was fixed in August](https://github.com/bitdump/BLHeli/issues/234#issuecomment-321849304) and said would release once final:
> @sskaug - Updated code is now on github, as Rev16.62. As this is only test code, I would not recommend to publish with the Suite or the Configurator. When everything tests fine, we will release a Rev16.7.

I [chimed in on Sep 17](https://github.com/bitdump/BLHeli/issues/234#issuecomment-330103003) to indicate the issue was still not fixed on 16.67-dshot.

There was further discussion, and then [@sskaug said they would go forward with changing beeper_dshot_beacon_tone to execute at "beacon" strength](https://github.com/bitdump/BLHeli/issues/234#issuecomment-330438499) as set it BLheli Suite:

> @sskaug - Great to have settled this now, We'll proceed with the release of Rev32.2 then.
And a good point you make on clarifying which volume is used when. Renaming controls in BLHeli can cause a lot of confusion ;). I propose we clarify this in the manual, and also in the mouse hover popup info of BLHeliSuite32.

...[and would also change the volume strength control in the UI from "_beacon_" to "_Beacon/Signal Volume_".](https://github.com/bitdump/BLHeli/issues/234#issuecomment-330596051)

Are these changes be implemented or still pending release ?

the dshot beacon commands as implemented in _16.67-dshot_ do not execute at "_Beacon/Signal Volume_" but instead at "_Startup Beep Volume_" so have limited usefulness on betaflight unless we crank the "_Startup Beep Volume_" way up high, which then also makes the startup tones rediculously loud and obnoxious.

Just trying to figure out if this is still pending after all these months, or is this a regression that was missed ?

Thanks !

-=dave

sskaug, 2017-12-11:
Have you tested the released Rev16.7 code?

fiveangle, 2017-12-11:
Perhaps this is a knowledge problem ?  In _16.67-dshot_, the startup tone sequence is: **3-[progression number of beeps depending on protocol identification]-[arming tones]**:

> ESC Signal Identification Tones Decoded:
> | PWM - 1 beep
> | Oneshot125 - 2 beeps
> | Oneshot42 - 3 beeps
> | Multishot - 4 beeps
> | Dshot - 5 beeps
> Arming Tones Decoded:
> | Single Arming Tone - Dshot150, Multishot, Oneshot, Oneshot42, PWM
> | 2 Step Arming Tone - Dshot300
> | 3 Step Arming Tone - Dshot600

Is this no longer true ?

Running 16.7 I received the 3-1-1 startup sequence, which would seem to indicate PWM detected so I assumed 16.7 was some other release not intended for dshot users (like has been the case for some time, with the "-_dshot_" suffixed FW names). 

Flashing back to _16.67-dshot_ resumed standard Dshot600 3-5-3 tone sequence.  

If 16.7 should work, then I can rename this issue to be the protocol detection problem.  If you say the beep issue is fixed in 16.7, that's great.  Once I get it working I will test and re-open if not resolved.

Thanks,

-=dave

sskaug, 2017-12-11:
The regular BLHeli_S (which Rev16.7 is) never have had signal identification tones.
Signal detection is done automatically and transparently.

Thanks,
Steffen

closed 2023-11-14 by fiveangle