bitdump/BLHeli #283: Turtle mode no longer working on kiss fc with blheli 32

Drewlrox, 2017-12-29:
With the latest kiss fc v2 update 1.3rc32 activating turtle mode just makes the escs beep, On rc30 turtle mode works perfectly. I have blheli 32.3 flashed to 2 different quads with 2 different blheli 32 escs showing this behavior. I understand this isnt really a blheli issue as flyduino made this change but I have asked and it seems like they are not interested in going back to the old method of activating turtle mode so I am asking here in hopes somebody knows why this is happening and if there is any fix for it. It would be nice to be able to use turtle mode with kiss again.

4712, 2017-12-30:
@Drewlrox  As far as I could test KISSFCV2F7-1.3RC32s, it seems to be an issue in combination with "Advanced... Buzzer emulation... Emulate with motors". After enable this option, the motors do no longer reverse the direction in Turtle mode.

[image]

JoKnutsel, 2018-01-10:
it's is known that turtle mode only works on Dshot command AND emulate beep on esc needs to be DISABLED.
on RC30 V1 and V2 fc

closed 2018-03-04 by Drewlrox