bitdump/BLHeli #284: Really missiong low voltage cutoff on multi. Anyway chance to bring it back?

pompecukor, 2017-12-31:
Hi,

The Low voltage cutoff is there in the plane/main, but not in multi.
I use multi for plane (fixedwing) as I need bi-direction (reverse trusting for precision landing speed control).
However the LVC is not available in multi.
Anyway to make is show? As it is in the program, I am guessing it is just a simply code/hack to make it happen. Maybe even just a Hex mod or something. Like have is hard coded to 3.0v (which is what we need), instead of "off" which is what it is at now.
This would be such a big help for us.
Thank you
Alex

sskaug, 2017-12-31:
There are no plans for adding this to BLHeli MULTI or BLHeli_S.
But it is available in BLHeli_32.
Thanks,
Steffen

pompecukor, 2017-12-31:
I bought the BLHeli_32 for just this reason. but even there I only see this on the "MAIN" page. Main as in main rotor. Which has no bidirectional throttle.
Biderectional is only in tail and multi, however those do not have the LVC.

sskaug, 2017-12-31:
As most ESCs are intended for multirotors, not all ESCs have voltage detection circuitry. The choice of whether to include this or not, is up to each manufacturer. Which ESC do you have?
But all BLHeli_32 ESCs support bidirectional mode. And for BLHeli_32, there is only one code (not MAIN/MULTI/TAIL).
Ar you sure you have downloaded BLHeliSuite32?

pompecukor, 2017-12-31:
I will check.

pompecukor, 2017-12-31:
You are so right sir. I downloaded the wrong version. Interesting that BlHeli site has both 16 and 32 on their download page.
Thank you for the help sskaug.

pompecukor, 2017-12-31:
[image]
Well no LVC here.
I don't have the ESC connected yet, but should it not be there anyway as the other options?

4712, 2017-12-31:
@pompecukor You have no valid ESC data loaded,...
[image]

pompecukor, 2017-12-31:
Wonderful thank you. I got Wraith from fpvmodel, 50A. Just with my technician at the moment. So that would have it I hope. Thank you.