bitdump/BLHeli #285: Gs35 4 in 1

ajharrop, 2018-01-08:
Can’t connect to blheli 32 configurator, I’m using Skitzo flight controller.
I get error code 2

4712, 2018-01-08:
Means, the system cannot find the specified serial port. 
Most likely wrong serial port is selected.

ajharrop, 2018-01-08:
ive tried all the ports on my computer but it still doesn’t work

4712, 2018-01-08:
How do you connect to configure the FC? Is it Raceflight?

ajharrop, 2018-01-08:
Raceflight one

4712, 2018-01-08:
Sorry, for BLHeliSuite it is not possible to connect using RF.
Raceflight does not support 4way-interface pass through, like Betaflight or Cleanflight.

ajharrop, 2018-01-08:
How can I connect the gs35 4 in 1 to blheli with the usb?
I’ve seen videos for the single esc’s but not the 4 in 1

ajharrop, 2018-01-08:
Without the usb*

4712, 2018-01-08:
4in1 is 4 ESC on one board. They still act as single ESC, means the have all one separate input wire to connect to the FC or another interface.
I found some hints that RF has a mode to activate a 1-wire pass through mode for each single ESC channel.

ajharrop, 2018-01-08:
Could you please share? I can’t find anything.
When I use 1wire read nothing happens

4712, 2018-01-08:
Ok, I thought, there is a 1-wire pass through mode (like in BF), but I guess, I was wrong..

mikeller, 2018-01-08:
There is a simple solution: Just use Betaflight. ;-)

ajharrop, 2018-01-08:
I prefer raceflight, I could use betaflight to configure my esc’s and then reflash rf1. That should work in theory as when you flash it’s only the fc right?

mikeller, 2018-01-09:
Yeah, that would work - use the REVOLT target in Betaflight.