bitdump/BLHeli #286: update 32.31 beta

JoKnutsel, 2018-01-10:
changes are : 
Added programmable current sensor calibration (not shown in Blheli suite beta)
Added programmable nondamped mode OK
Added support for ProShot input signal OK
Added input signal status reporting (not shown in Blheli suite beta, or is it the log info ? )
Added programmable startup music (not YET shown in Blheli suite beta)

where you can calibrate the current sensor ?

4712, 2018-01-10:
You need to use the BLHeliSuite32Betaas shown in the Readme.md file [/BLHeli/blob/master/BLHeli_32 ARM/README.md](https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/README.md). The option is only shown, if a valid setup of an ESC with current sensor is loaded.  

[image]

JoKnutsel, 2018-01-10:
i use the beta suite. and have dalrc engine 40a 4in1 with current sensor.
will check tonight again

4712, 2018-01-10:
The DALRC Engine 4in1 does support a single current sensor per all ESC together, which cannot be handled by BLHeli_32 and therefore will not show up as a feature in BLHeliSuite32. It must be handled by the FC as external current sensor and calibration must be done in the FC firmware.

JoKnutsel, 2018-01-10:
to bad kiss v2 fc doesnt allow current sensor calibration