bitdump/BLHeli #288: Missing closed loop mode in BLHeli_S and recent use case

Dronek, 2018-01-16:
I'm wondering why closed loop mode is available in BLHeli but not in BLHeli_S. Are there technical reasons or was it just not used by the majority?

Nowadays there is an interesting use case for closed loop mode: LiIon batteries (a popular choice for long range or long flight time applications) span a relatively large voltage range. For example, a 4s LiIon battery spans a voltage range from almost 16.8 V (equivalent to a fully charged 4s LiPo) down to 10 V when empty and under load (equivalent to an empty 3s LiPo). Without a closed loop mode this voltage range translates to a wide range of thrust at constant throttle. When controlling (just) the throttle, the pilot has to account for the loss of thrust with decreasing voltage and has to manually adjust the throttle to keep hovering (multirotor) or cruising at the same speed / prevent aircraft from stalling (fixed wing). 

It would be much more elegant to have control over RPM (thrust) in such a use case plus it would make life easier for autopilot applications.

Is there any chance to have closed loop mode (or something similar) added to BLHeli_S ?

sskaug, 2018-01-16:
Closed loop mode was added to BLHeli, since it already had the governor functionality for helicopter main motor applications. But as far as I know, closed loop for multicopter was hardly ever used. Did you use it?

A complication with multirotors is that it is hard to calibrate max throttle. Also, of course, max power will not be at max throttle, except for at a very specific voltage (if at all, that is). There are no plans for closed loop mode for BLHeli_S or BLHeli_32 at the moment.

Dronek, 2018-01-17:
> But as far as I know, closed loop for multicopter was hardly ever used. Did you use it?

I had no reason to use it with BLHeli on FPV miniquads at that time. Now with fixed wing aircraft and especially LiIon batteries it is a different story when it comes to efficient flight and being able to control the actual thrust (e.g. keeping it constant over a certain voltage range) rather than just the throttle. 

> A complication with multirotors is that it is hard to calibrate max throttle. Also, of course, max power will not be at max throttle, except for at a very specific voltage (if at all, that is). 

In my mentioned use case it is often not of interest to push power limits so max throttle or max power related topics should be a minor issue.

> There are no plans for closed loop mode for BLHeli_S or BLHeli_32 at the moment.

I understand that there is not a huge demand for closed loop mode. With ESC telemetry in BLHeli32 the task of keeping RPM constant in spite of a gradual voltage drop could possibly be handled by the FC, which again results in a closed loop with a larger loop, I guess.

Just for the sake of completeness, the background of this discussion: [https://github.com/iNavFlight/inav/issues/2649](url)

jcg1541, 2019-12-02:
> Closed loop mode was added to BLHeli, since it already had the governor functionality for helicopter main motor applications. But as far as I know, closed loop for multicopter was hardly ever used. Did you use it?
I use it every day, "Closed Loop Mode: On" in my page screenshot
https://nocomputerbutphone.blogspot.com/2018/09/converged-iot-drone-build-notes.html#esc_rpm
. I hope BLHeli_32 can bring back the PID loop. 
I looked at the asm code of the PID loop of original BLHeli version 14.x . It is very interesting to see the raw machine code in action.