bitdump/BLHeli #289: Motors won't Arm using OneShot125

GS-AutoTech, 2018-01-24:
I have 2 builds in progress. Both running Airbot F7V2 FC's & iNAV.  In the first quad I installed 4 BetaFlight 35A BLHeli32bit ESC's. Flashed board to BF so I could use passthru BL32 Suite. Configured and calibrated ESC's,  flashed back to iNAV, set OneShot125 and it flies perfect.
On the second build I installed 4 Airbot Wraith32's. Performed all the same procedures as the first build but ESC's would not arm. After a few hours of diag I tried the D-shot config in BF. ESC's arm & motors spinning fine.. Any time I select anything other than D-shot, it won't Arm.
I need OneShot125 to use iNAV that the quad is built for.  Is it possible these new Wraith32's don't support OneShot125? 
I've tried every combination I could think of. Even flashed Wraith32's back to earlier BL32 FW. 
I hope.there is something simple that I missed.
Any help would be greatly appreciated. 
Thanks for all.your hard work guys!

George @
GS AutoTech

sskaug, 2018-01-24:
You need to raise Minimum Throttle in BLHeliSuite.
Or reduce min command in the FC.

GS-AutoTech, 2018-01-24:
Oh boy, I hoped the issue might be that simple.. 
I will try that right away.
Thanks you for your rapid response & all of your hard work. 

GS

GS-AutoTech, 2018-01-24:
I reset minimum throttle & it worked perfectly!

Thank you very much!

GS