bitdump/BLHeli #290: additional Modes for Plane and Car

Hrumque, 2018-01-28:
I think that additional options (work modes) outside of Main / Tail / Multi would be useful.

1.  Aircraft (especially useful for a glider with a folding propeller) - where throttle regulates the motor control in 3 ranges

- braking (active or passive)
- free wheel
- acceleration (with or without governor option).

 It would allow to fly with the propeller extended (and use the surface of the propeller as an aerodynamic brake) and after braking the engine - make the propeller fold.

2.. The car. Here, of course, two-way rotation is needed, but the usual choice of "bidirectional" (and braking by dampedlight, or no braking) does not give full control. A 5-band job would be useful:
- backwards
- free wheel (free turning)
- braking @ throttle = center = eg. 1500 (with regulated braking power via pwm for mosfets?)
- free wheel (free rolling)
- forward.
And the "throttle response speed" would be useful (to reduce the torque impact on the drive system)

3. the mode which I call "drill". Mode without arming, after switching on the power, ESC only checks if it is in programming mode via pwm with blheli-suite, and if not - it simply turns on the RPM (one or two-way, with speed and direction depending on PWM). Useful for modifying, for example, power tools for a brushless motor :), where arming every time you turn it does not make sense, we just want the motor to spin like a DC motor.

Currently, there is a lot of inexpensive and very good ESC with BlHeli_S firmware (3.99$ for 28A Esc with BB2 cpu is good option), which would be suitable for many applications, but the lack of basic functions in blheli/bheli_s means that you must use historic esc built on '51 and simonK-like firmware

voneiden, 2018-01-28:
> lack of basic functions in blheli

To be fair, those are neither basic or relevant functions for a heli.

mikeller, 2018-01-28:
> To be fair, those are neither basic or relevant functions for a heli.

Good point. Maybe BLHeli should be renamed into 'BLRc' or similar. Helis seem to be a just as exotic use case for this hardware as gliders or cars.

Hrumque, 2018-01-29:
The name is just a name. Also this project could be called very-good-firmware-for-esc or ESCoMATIC2000 - it's not the most important thing, it's just that you're doing great firmware for ESCs, which could be used in many areas. I think it would be good to "attack all markets" if you can?

cturn, 2018-01-29:
Well, you make a decent case for expanding the user base to DIY motor swaps with something like the drill mode.  brushless powerwheels mod?  I could see it being financially worth while for them to add something like that to Blheli32.  But so far as I am aware, no ones is currently actively developing the open source versions.  However your in luck if your wanting the “drill mode” thing.  I modified a version of blheli_s for a friend that does basically exactly what you describe under “drill mode” just shoot me a msg if you want It.

Jarret-MIS, 2018-02-05:
I've tried CC3D FCs and fixed-wing specific ESCs for my planes.  They are far to large, and at great cost and time to just throw anything in them.  No BLHeli ESC would cut it in most cases.  Not all, but most planes over 50" wingspan using Outrunners in the 5056 range, or Axi's 4130/16 motors.