bitdump/BLHeli #293: BLHeli32 - Dumped mode can potentially destroy the ESC

lakotamm, 2018-02-04:
The "Dumped" or active braking mode on BLHeli32 can burn the ESC if it is connected to a source which has a high internal impedance or can't handle regenerative currents.

During my testing, ESC (Wraith32 V2) was able to generate over 27V while using 12V power supply. The peak voltage during some of the tests was over 34V. This voltage can seriously harm the ESC.

Suggestion:
- mention in the documentation, that ESC has to be connected to a source which can withstand a high regenerative current
- implement a simple testing procedure which would allow the user to test, whether his source is safe to use with the damped mode 
- allow disabling dumped mode

Example of the testing procedure:
Run a serious of tests when a motor spins up to a certain RPM and then brakes down. The ESC is monitoring the voltage of the input, and if it reaches a certain value (+5V over the supply line?, 25V supply voltage? 30% increase?) it informs the user that the use of the dumped mode can harm his quadcopter

[image]

CreeT6, 2018-02-08:
why do we not have MAIN firmware versions and no option to disable braking :((((((

sim-, 2018-02-08:
It works quite well, if the ESC is capable of measuring the input voltage, to break PWM if the input voltage exceeds a safe maximum. I did this on some stm32f3-based ESCs with good success (actually in hardware on that one, with a dac vs voltage fed to a comparator fed to one of the timer break channels). Maybe that would be possible here?

lakotamm, 2018-02-08:
VESCv6 has an option where you can set up maximum safe input voltage and I think that the voltage doesn't go over that value  This would be an option.

The HW solution to the issue is to use a shunt regulator on the input.
[image]
[image]

McGiverGim, 2019-02-24:
Sorry if it is a silly question, I don't know too much about electronics, but I think maybe I have suffered this problem.

Every time I go to fly my quad, I take off, I fly without problem until I try some "aggressive" maneuver, for example a roll. At this moment some ESCs die. 
- The first time were four Wraith 32 v1.1, all four died. No connection, no startup tone, no led.
- The second time were two Hakrc 35A, both from the side that were pushing the roll.
- The third time were another two Hakrc 35A, both from the rear part of the quad.

I have checked all connections of the FC with a multimeter, searching for some kind of short, but I can't find nothing.

The strange thing is that if this is a "real" problem, a lot of other people will suffer it and I have not found nobody else. :(

d4rken, 2019-12-23:
I think I've experienced this. On the workbench, without props, connected to a switching powersupply.

I was testing current draw via Betaflight motor control. When removing throttle, the motors stop instantly.

After a few runs one turning on Motor 1 would reboot the stack due to the ESC exceeding my powersupplies current limit. As far as I can tell, one MOSFET (part of the Motor 1 ESC) died (F55A II Pro).

LouDnl, 2020-05-28:
I have experienced this 3 times now with Hakrc 35A 32Bit escs. First time my quad just fell out of the sky during a powerloop. When testing the esc led would only light up briefly when plugging in de battery. Order 2 new escs and after rebuilding the quad one of the escs died during an on purpose bump landing while doing the first test hover. Exact same problem as before.
Third time was with the new spare esc on the bench. Hooked it up to 15 volts from my bench power supply to do some voltage measuring and it just died.
On the hakrc escs there's an st positive voltage regulator that outputs 12 volts that goes to an ldo regulator that makes 3.3v for the stm32 and fet driver. In between these two ic's exists a problem.
When applying power the 12v regulator doesnt put out 12v but only 2v which makes the ldo regulator not function properly. Applying 12v to the output of the regulator makes the ldo work properly. Removing the ldo makes the regulator work properly again, but testing the ldo separated from the escs shows it works just fine...
The high voltage feedback it dies from breaks something which I have yet to find.

McGiverGim, 2020-05-28:
In my case, it seems it has something to do with the ESC telemetry. After breaking a lot of ESCs, I disconnected it and since then the problem is gone. I checked several times the connections for a short and I didn't find any.

LouDnl, 2020-05-28:
> In my case, it seems it has something to do with the ESC telemetry. After breaking a lot of ESCs, I disconnected it and since then the problem is gone. I checked several times the connections for a short and I didn't find any.

I have that hooked up, good thing to test and start from. unfortunatly no working extra escs at the moment.

LouDnl, 2020-05-28:
The hakrc 35a esc uses a positive voltage regulator to give out a steady 12v to drive the onboard rgb led. And as far as i can see its the only thing that uses the 12v. There is an ldo regulator that creates 3.3v from the 12v to drive the stm32 and the fet driver. After removing the 12v regulator and feeding 5v to the old 12v output makes the esc work again and the led works too.
Have yet to test the 12v regulator if it works separeted from the esc. I have already tested the 3.3v reg and that works just fine.
Makes me wonder if its a design flaw.

LouDnl, 2020-05-28:
Just tested the 12v reg and 3.3v reg removed from the esc and hooked up to each other and they both work normally... strange

MarcinKoz, 2020-07-09:
A zastanawialiście się po co na zasilaniu esc przyłacza  się kondensator 470-1000uf na napięcie najlepiej 2 razy większe od zasilania.
Właśnie po to by esc podczas hamowania nie wprowadzał na zasilanie wysokich napięć i zakłóceń. Jak wiemy napięcie na kondensatorze nie może zmieniać się skokowo i kondensator jest filtrem. Każdy kto używa ścigaczy z dshot i damped mode ma zamontowane kondensatory i nie ma problemów.

haslinghuis, 2020-07-09:
@MarcinKoz please use English. Did you mean:
```
And you wondered why the 470-1000uf capacitor is connected to the esc power supply for a voltage preferably twice as high as the power supply.
Precisely so that during braking esc does not introduce high voltages and interference into the power supply. As we know, the voltage across the capacitor must not change abruptly and the capacitor is a filter. Everyone who uses chasers from dshot and damped mode has capacitors mounted and has no problems.
```
What are `chasers`?

LouDnl, 2020-09-29:
> @MarcinKoz please use English. Did you mean:
>
> ```
> And you wondered why the 470-1000uf capacitor is connected to the esc power supply for a voltage preferably
> twice as high as the power supply. [...]

I think the translation might be damped light?

> And you wondered why the 470-1000uf capacitor is connected to the ESC power supply for a voltage preferably 2 times higher than the power supply.
Precisely so that the esc during braking does not introduce high voltage and interference to the power supply. As we know, the voltage across the capacitor cannot change in steps and the capacitor is a filter. Everyone who uses dshot and damped mode speeders has capacitors installed and no problems.

haslinghuis, 2020-09-29:
According to the video use this settings:

Setting | Drones | Gliders | Comment
-- | -- | -- | --
Non Damped Mode | On | Off | On = Turns braking off
Brake on stop | Off | On | Second type of braking

https://www.youtube.com/watch?v=sg0qPBYRB60