bitdump/BLHeli #299: BLHeli_32 passthru does not work on RTFQ Omnibus F4 w/Robotdog ESCs

aowi7280, 2018-02-22:
Passthru did not work for 2 ESCs. I tried to reflash several times and it always failed. I was able to change/save settings on the 2 good ESCs. So I got replacements. I installed the 2 new ESCs and they also do not communicate properly. And now one of the other ESCs is no longer recognized by BLHeli_32. So 3 ESCs do not communicate, 1 does communicate. They run OK and I can fly, but I cannot change any of the settings because BLHeli_32 does not recognize them.

I was using Betaflight 3.2.4 in my flight controller. Then I upgraded to Betaflight 3.2.5 and the BLHeli_32 pass through still does not work right; using latest revision. I even tried the BLHeli Beta version, no good. Tried different USB ports on PC, no good.

4712, 2018-02-22:
Did you connect all input ground wires from the ESCs to the FC board?
What protocol do you use for the ESC, DShot?

aowi7280, 2018-02-22:
Yes, I did.  I had been using Dshot, but someone suggested I try Oneshot125. That did not work. Neither did PWM.

4712, 2018-02-22:
Could you please post the FC controller settings file and the cli dump file? Maybe I can reproduce the issues then.
Did you try to change the FC outputs for the ESCs?

aowi7280, 2018-02-23:
ESC#4 was the only one reading so I swapped the connector with #3 at the FC. It would not connect at all with Oneshot selected in the FC. I selected Dshot 1200, rebooted and it connected to all 4 ESCs. I put the 2 connectors back where they belong and it still works fine. I could even save settings.
Not sure what you mean by FC settings file. But here is the cli dump.
# dump

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.2.5 Feb 11 2018 / 00:49:36 (6e69ff00c) MSP API: 1.36

# name
name Androne

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 C09
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 C08
[... 572 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_rate_yaw = 85
set rc_expo = 10
set rc_expo_yaw = 10
set thr_mid = 50
set thr_expo = 0
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 75
set tpa_rate = 10
set tpa_breakpoint = 1650

aowi7280, 2018-02-23:
I think I figured it out. The FC cover, made of conductive carbon fiber, was resting on the PWM output pins. I thought I had space. I still need to fix and reassemble. But thanks for the help.

mikeller, 2018-02-23:
Ouch. Hope you did not fry your electronics! Kapton tape (ebay has it) is a good and durable solution for this kind of insulation. If this is fixed, can you please close the issue?

aowi7280, 2018-02-23:
All fixed with 2mm longer standoffs and some hot glue for insurance. Guess the carbon had enough resistance not to fry anything.

closed 2018-02-23 by aowi7280