bitdump/BLHeli #300: Arduino UNO Pin 8 kills my ESCs

mbr1989, 2018-02-23:
XRotor 20A / Arduino UNO / 4wArduino_m328P_16_MULTIv16400.hex

https://cdn.discordapp.com/attachments/194896601133744129/416534526450204672/unknown.png

Already changed the Arduino UNO / Amazon -> Arduino CC

4712, 2018-02-23:
@mbr1989 
Maybe some more info would help to understand, what happens...
What did you so before?
Does it happen at other Pins than #8?
Any log file?

mbr1989, 2018-02-23:
Nope, all other Pins are fine ... only Pin 8 kills the ESCs

mbr1989, 2018-02-23:
23.02.2018 10:59:11: "Write Setup Section Failed" [Target ESC: Test - XRotor20A - Rev. 14.9 - Multi]

mbr1989, 2018-02-23:
> What did you before?

Well, I flashed my ESCs through C2 Interface

4712, 2018-02-23:
Ok, I will look at it ASAP.

4712, 2018-02-23:
Sorry, I tried hard using AIKON SEFM 30A at pin 8 vs. 7, but could not find any difference between the Pins with my UNO board. All are working well here. 
Also, from looking at the UNO schema, I found no reason for any issue at pin 8 so far.
But maybe your board is different from mine. 
Please double check the wiring of Pin 8 to the ESC, and check the signal with a scope.

mbr1989, 2018-02-23:
Arduino Uno Rev3
SKU 8058333490090

mbr1989, 2018-02-23:
https://www.amazon.de/gp/product/B008GRTSV6/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

mbr1989, 2018-02-23:
[image]

4712, 2018-02-23:
OK, now I found the issue...

4712, 2018-02-23:
@mbr1989 
Thanks for the finding!!!
I need to look into the code, seems to be a timing issues in the flash erase command.
Meanwhile please try the file from BLHeliSuite32 for the UNO.
[4wArduino_m328P_16_MULTIv20002.zip](https://github.com/bitdump/BLHeli/files/1753320/4wArduino_m328P_16_MULTIv20002.zip)

mbr1989, 2018-02-23:
My test ESC finally fucked up ... no chance to flash it over C2 any more .... ^^

4712, 2018-02-23:
There is no need to reflash over C2 unless the bootloader gets corrupted. If the "u" is shown with a bootloader info (d), the ESC should be ready for flashing over bootloader and 4way-interface/1-wire.

mbr1989, 2018-02-23:
Nope, didn't worked for me ^^

4712, 2018-02-23:
Yes, sorry, not in with Arduino UNO Pin D8, but e.g. with Pin D9.
[image]
This shows succeeded flash of "u" marked XRotor 20A with UNO 16400 hex after a prior page erase failure.

mbr1989, 2018-02-24:
> Yes, sorry, not in with Arduino UNO Pin D8, but e.g. with Pin D9.

 Puhhhhh, not sure about that ... I'll keep that in mind

4712, 2018-02-25:
Fixed with BLHeliSuite16.7.14.9.01

mbr1989, 2018-02-25:
https://cdn.discordapp.com/attachments/194898549694005250/417393634367897600/BLHeli_X1_-_XRotor20A_-_Rev._14.9_-_Multi_180223.ixi

mbr1989, 2018-02-25:
:030000000219FDE5
:03000B000200AA46
:03002B0002018748
:03005B000203415C
:0300730002032E57
:1000800002030406080C10182030406080040608A3
:100090000C10182030406080A0C00003070F1F0D17
:1000A0000D04050D050305030302C2AF306308C24A
:1000B0006385798AD2AF32C0D0C0E0206221E525C5
:1000C0006016E473E525F4C3334009758A00D263F2
:1000D000F57901D6F58AD262D0E0D0D0D2AF32C362
:1000E000E526334009758A00D263F57901F0F58A77
:1000F000C262E526F4602F30682520720DD0E0D072
:10010000D0C292C296C294D2AF32C292C296C29468
:10011000206F0C206509741AD5E0FDE57B4290D074
:10012000E0D0D0D2AF32758A00D263757900C28D2B
:10013000D262211F01C4306805206502D29201C439
:10014000306805206502D29401C43068052065023C
:10015000D29601C4C29330680A206507741CD5E0AA
:10016000FDD29201C4C29530680A206507741CD57F
:10017000E0FDD29401C4C29730680A206507741C60
:10018000D5E0FDD29601C4C2AFC2AD53E6EFC0D0F8
:10019000C0E0D2D3D2AFE57A6009E5776005757724
:1001A0000041DB757701C2CEE52A600720744F1548
:1001B0002A21FE7800790020742EE580307E01F43B
:1001C00030E50278FF53DACF207E0343DA20307E19
:1001D0000343DA10C2D8C271E580307E01F430E505
:1001E0000279FFC3E89970CB306103752A18207437
:1001F00003752A18885CD270E86003755F0020746C
:100200000DE52B6004152B410E43DA01C2D8207096
:1002100002417DA85C206102C2702074367982B7E9
:100220000431C3E894F0400374F0F8E8C4540F2894
:10023000F87984B703024153C313C31387212008FD
:1002400002C313200A06C3C898F8415328F8500384
:1002500074FFF88822E52D54066022206B1FE536D6
:10026000C333C333F8E564C313C313F9C395224002
:10027000028922E82522F52250037522FF7882B6F2
:100280000459852224A824C3E52495614002A8616D
:10029000C3E895634002A8638825E566700241CEF5
:1002A000C3E8AA669A5003E8FAE4F9EAC333FBE527
:1002B00068F45B29400D2567F8E567600215674023
:1002C0000B41CE0AC3E5679A5002056778FF88267E
:1002D000C26FC3E52694F84002D26F20CF0AD0E067
:1002E000D0D043E610D2AD32C2CF053AE57A6009EC
:1002F000E578600575780061247578017801E52A54
:100300006005307402152AC3E55C94014005755FF1
:10031000006124756000756900E55F2401F55F5098
:1003200003755FFFD0E0D0D043E610D2AD32C2AF4C
:1003300053E67F7592FA7593FFC260759104D2AF50
:1003400032C2AF53E6EFC2ADC0D0C0E0C0F0D2D3EE
:10035000D2AFA8FBA9FCE57A6007C3E913F9E8135B
:10036000F8C2D830710261B353DACF207E0343DA8A
:1003700010307E0343DA20D271E580307E01F42014
:10038000E502618A88278928A1E053DACF207E031D
:1003900043DA20307E0343DA10C2D8C271307402CF
:1003A000A1CC7800E580307E01F430E502A1CC8854
:1003B0005CA1B453DACF207E0343DA20307E0343BE
:1003C000DA10C271206102818C53DACF207E0343A0
:1003D000DA10307E0343DA20C2D8D27188578958A8
:1003E000C3E89555F8E99556F97B007E087A00C375
:1003F000E8948CE994005005755B00817E88217833
:10040000A2E6A8216055C3E894C8E994005008E426
:10041000D2E4FB7A0A8159C3E89468E99401500850
:10042000E4D2E3FB7A0F8159C3E894D0E9940250F7
:1004300008E4D2E2FB7A1E8159C3E894A0E994054E
:100440005008E4D2E1FB7A3C8159C3E89498E994DE
:10045000085008E4D2E0FB7A787E00C3E89559FCA6
:10046000E9955AFD30E70AECF42401FCEDF4340080
:10047000FD755B00C3EC9AED9E5003755B018859D6
:10048000895A8557558558567802A1B4C3E89527EF
[... 287 lines cut ...]
:10168200121704E520B4AA030216A2757CA5757D83
:10169200F1120FA41216C2757C00757D000216C1EC
:1016A200901A0378807B0A121704A308DBF9901AB8
:1016B2000F788C7B19121704A308DBF9901A2822E1
:1016C200C2AF121747121724901A00740E1217098C
:1016D200A37409121709A37415121709901A037833
:1016E200807B0A121708A308DBF9901A0F788C7B0B
:1016F20019121708A308DBF912175A121738901A91
:101702002822E493F622E6438F01538FFDFFC3E5BF
:1017120083941C400122EF857CB7857DB7F0538FFF
:10172200FE22438F02438F01857CB7857DB7901AD5
:101732000DF0538FFC22901A0D7455F109901A0E78
:1017420074AAF109227A3079D07820901A40F104F3
:10175200E520F709A3DAF7227A3079D0901A40E728
:10176200F10909A3DAF922AB74A875BB01027980E9
:10177200BB02027981BB03027982BB04027984BB7A
:1017820005027986BB06027992BB07027987BB08FC
:101792000279A3BB0902799CBB0A027988BB0B02BE
:1017A2007989E8F7227C0512060BDCFB22C2AF1214
:1017B200061B1206221206291206301205FC120618
:1017C2001B1206221206291206301205FC12061BF3
:1017D200120622120629120630D2AF22C2AF120618
:1017E2003012062912062212061B1205FC120630BE
:0F17F20012062912062212061B1205FC120630DF
:00000001FF

mbr1989, 2018-02-25:
``:030000000219FDE5
:03000B000200AA46
:03002B0002018748
:03005B000203415C
:0300730002032E57
:1000800002030406080C10182030406080040608A3
:100090000C10182030406080A0C00003070F1F0D17
:1000A0000D04050D050305030302C2AF306308C24A
:1000B0006385798AD2AF32C0D0C0E0206221E525C5
:1000C0006016E473E525F4C3334009758A00D263F2
:1000D000F57901D6F58AD262D0E0D0D0D2AF32C362
:1000E000E526334009758A00D263F57901F0F58A77
:1000F000C262E526F4602F30682520720DD0E0D072
:10010000D0C292C296C294D2AF32C292C296C29468
:10011000206F0C206509741AD5E0FDE57B4290D074
:10012000E0D0D0D2AF32758A00D263757900C28D2B
:10013000D262211F01C4306805206502D29201C439
:10014000306805206502D29401C43068052065023C
:10015000D29601C4C29330680A206507741CD5E0AA
:10016000FDD29201C4C29530680A206507741CD57F
:10017000E0FDD29401C4C29730680A206507741C60
:10018000D5E0FDD29601C4C2AFC2AD53E6EFC0D0F8
:10019000C0E0D2D3D2AFE57A6009E5776005757724
:1001A0000041DB757701C2CEE52A600720744F1548
:1001B0002A21FE7800790020742EE580307E01F43B
:1001C00030E50278FF53DACF207E0343DA20307E19
:1001D0000343DA10C2D8C271E580307E01F430E505
:1001E0000279FFC3E89970CB306103752A18207437
:1001F00003752A18885CD270E86003755F0020746C
:100200000DE52B6004152B410E43DA01C2D8207096
:1002100002417DA85C206102C2702074367982B7E9
:100220000431C3E894F0400374F0F8E8C4540F2894
:10023000F87984B703024153C313C31387212008FD
:1002400002C313200A06C3C898F8415328F8500384
:1002500074FFF88822E52D54066022206B1FE536D6
:10026000C333C333F8E564C313C313F9C395224002
:10027000028922E82522F52250037522FF7882B6F2
:100280000459852224A824C3E52495614002A8616D
:10029000C3E895634002A8638825E566700241CEF5
:1002A000C3E8AA669A5003E8FAE4F9EAC333FBE527
:1002B00068F45B29400D2567F8E567600215674023
:1002C0000B41CE0AC3E5679A5002056778FF88267E
:1002D000C26FC3E52694F84002D26F20CF0AD0E067
:1002E000D0D043E610D2AD32C2CF053AE57A6009EC
:1002F000E578600575780061247578017801E52A54
:100300006005307402152AC3E55C94014005755FF1
:10031000006124756000756900E55F2401F55F5098
:1003200003755FFFD0E0D0D043E610D2AD32C2AF4C
:1003300053E67F7592FA7593FFC260759104D2AF50
:1003400032C2AF53E6EFC2ADC0D0C0E0C0F0D2D3EE
:10035000D2AFA8FBA9FCE57A6007C3E913F9E8135B
:10036000F8C2D830710261B353DACF207E0343DA8A
:1003700010307E0343DA20D271E580307E01F42014
:10038000E502618A88278928A1E053DACF207E031D
:1003900043DA20307E0343DA10C2D8C271307402CF
:1003A000A1CC7800E580307E01F430E502A1CC8854
:1003B0005CA1B453DACF207E0343DA20307E0343BE
:1003C000DA10C271206102818C53DACF207E0343A0
:1003D000DA10307E0343DA20C2D8D27188578958A8
:1003E000C3E89555F8E99556F97B007E087A00C375
:1003F000E8948CE994005005755B00817E88217833
:10040000A2E6A8216055C3E894C8E994005008E426
:10041000D2E4FB7A0A8159C3E89468E99401500850
:10042000E4D2E3FB7A0F8159C3E894D0E9940250F7
:1004300008E4D2E2FB7A1E8159C3E894A0E994054E
:100440005008E4D2E1FB7A3C8159C3E89498E994DE
:10045000085008E4D2E0FB7A787E00C3E89559FCA6
:10046000E9955AFD30E70AECF42401FCEDF4340080
:10047000FD755B00C3EC9AED9E5003755B018859D6
:10048000895A8557558558567802A1B4C3E89527EF
[... 287 lines cut ...]
:10168200121704E520B4AA030216A2757CA5757D83
:10169200F1120FA41216C2757C00757D000216C1EC
:1016A200901A0378807B0A121704A308DBF9901AB8
:1016B2000F788C7B19121704A308DBF9901A2822E1
:1016C200C2AF121747121724901A00740E1217098C
:1016D200A37409121709A37415121709901A037833
:1016E200807B0A121708A308DBF9901A0F788C7B0B
:1016F20019121708A308DBF912175A121738901A91
:101702002822E493F622E6438F01538FFDFFC3E5BF
:1017120083941C400122EF857CB7857DB7F0538FFF
:10172200FE22438F02438F01857CB7857DB7901AD5
:101732000DF0538FFC22901A0D7455F109901A0E78
:1017420074AAF109227A3079D07820901A40F104F3
:10175200E520F709A3DAF7227A3079D0901A40E728
:10176200F10909A3DAF922AB74A875BB01027980E9
:10177200BB02027981BB03027982BB04027984BB7A
:1017820005027986BB06027992BB07027987BB08FC
:101792000279A3BB0902799CBB0A027988BB0B02BE
:1017A2007989E8F7227C0512060BDCFB22C2AF1214
:1017B200061B1206221206291206301205FC120618
:1017C2001B1206221206291206301205FC12061BF3
:1017D200120622120629120630D2AF22C2AF120618
:1017E2003012062912062212061B1205FC120630BE
:0F17F20012062912062212061B1205FC120630DF
:00000001FF

mbr1989, 2018-02-25:
https://cdn.discordapp.com/attachments/194898549694005250/417394905308987402/xr20a.hex

https://cdn.discordapp.com/attachments/194898549694005250/417394902976954378/xr20a_2.hex

4712, 2018-02-26:
This means, you got them revived?
Great! Thanks for the info.