bitdump/BLHeli #310: Blheli32 disarms/disables throttle control when rapidly decelerating over PWM

anton-rothwell, 2018-06-07:
ESC is Wraith32 Metal 60A V2.
Tested on BlHeli32 version 32.3 and 32.4

Steps to Repro:
1. Configure BLHeli with all failsafes off
2. Attach servo tester to ESC
3. Attach suitable motor to ESC
4. Power on
5. Increase throttle to 60%
6. Rapidly drop the throttle to 30% (in less than 1 second)

ESC will beep twice and throttle control will become disabled.  Lowering throttle to 0 causes ESC to beep once and normal throttle control will resume until a rapid decel is caused again.

Using firmware 32.4 and turning off damped light stops this from happening.  Only tested with PWM via a servo tester and PX4 based autopilot.

sskaug, 2018-06-07:
When you say that the ESC beeps twice, that generally means that it loses valid input signal for a while, then beeps twice as it recovers a valid signal. Maybe there is noise on your input signal when the motor does active braking (throttle going form 60% to 30% with damped light on)?
You could try the Rev32.42 test code that we just posted, maybe the input signal noise filter helps in your case.

Zuldan, 2018-06-07:
"You could try the Rev32.42 test code that we just posted".

Is it possible to put the releases on the Releases page on Github? Maybe with release notes. Downloads will be faster for everyone.

anton-rothwell, 2018-06-08:
Hi, 
Thanks for getting back.  I've tried Rev32.42 this morning and the problem still exists, possibly slightly less so - it seemed harder to reproduce and needed a bigger throttle delta but it was still there.  We have other non blheli32 ESCs (DYS 30A) with damping on where this has never been a problem.  So it could be unique to the Wraith32.