bitdump/BLHeli #311: Proshot doesn't always inititalize on all 4 escs with 16khz refresh rate (worse on 32khz).

flint723, 2018-06-10:
Seen it a few times myself where only 3, or 2 escs arm (out of 4) with Proshot.  Using matek f405 at 32khz gyro/16khz pid/proshot1000.  Also @fujin has seen it on his 32khz/16khz setup. 

I want to use Proshot because of the the FC cpu usage gains + dshot commands working, so would be nice if this worked.

Thanks,
Flint

Zuldan, 2018-06-10:
I'm getting the exact same issue on my Spedex GS35 with Proshot1000.

sskaug, 2018-06-10:
Thanks for reporting, I'll look into it.

sskaug, 2018-06-14:
I have now posted a Rev32.43 test code on github, where this shall be fixed.

flint723, 2018-06-15:
Seems to work, at least didn't have a problem yet initializing all escs on 32khz.  On the Aikon AK32.  Didn't get to try the Wraiths yet.

MReFPV, 2018-06-17:
Does proshot works on 32khz pid loop?

MReFPV, 2018-06-17:
Sounds good! Thx

HotwiresFPV, 2018-06-18:
I got a desync running proshot1000 Rev32.4.3 resulted a crash. also there's an instance where the motors armed but not throttle response.

MReFPV, 2018-06-18:
Thanks for reporting, so I'll stay away. Dshot1200 16/16 works w/o problems on helio.

Zuldan, 2018-06-18:
@HotwiresFPV, I know about 4 people who have updated to 32.43 to use Proshot1000 and they've had zero issues. 

Are you running stock BLHeli settings? Is your PWM Freq at 48Khz?

HotwiresFPV, 2018-06-18:
Yes I'm running stock at 48khz

Zuldan, 2018-06-19:
I've just seen another 3 people on Slack upgrade to Rev32.43 and have had no issues with Proshot1000. @HotwiresFPV maybe re-flash your ESC's and FC, leave default settings and see what happens.

sskaug, 2018-07-06:
Proshot detection shall now be reliable, in the published Rev32.5 code

closed 2018-07-06 by flint723