bitdump/BLHeli #313: Flash different target to STM

liquidacid, 2018-06-12:
Hello,

Im currently in the process of repairing a BLHeli32 board. Typical issue with broken STM32. I'd replace it from a burned board but with different target. I would like to know how to flash a different HEX file to my STM32 chip since BlHelisuite32 doesn't seem to let me do such things.

Thanks!

sskaug, 2018-06-12:
Unfortunately such operations are not supported by BLHeli_32. The bootloader is locked to a target, and we do not support changing bootloader.

liquidacid, 2018-06-12:
Yea found it out already. I'd send you an email about this. Please let me know your response :).

mikeller, 2018-06-13:
@liquidacid: What you can do is replace the MCUs on the flight controllers for new and unlocked ones - it's a hassle, but commitment to open source seems to be only skin deep in the ESC area at the moment.

closed 2018-06-24 by sskaug

iggut, 2019-09-20:
> Unfortunately such operations are not supported by BLHeli_32. The bootloader is locked to a target, and we do not support changing bootloader.

did you ever figure out a way to do this? I repaired 3 of my broken 4in1 escs and since I had none of the same ESC I can not flash the right firmware without some sort of workaround. Please let me know by email if it should not be posted here

frizzle101101, 2020-11-26:
I am currently in the same situation, how do we flash new hex files...so if this is how it is and open source projects like blhelis/m will support flashing different models with no real advantage for 32 bit anymore i guess these are going in the trash, what a waste i can't believe