bitdump/BLHeli #314: Spedix GS30A not working with 32.3 or higher

supiiik, 2018-06-18:
Hi, I have 3 spedix GS30A (bootloader h) and 1 spedix GS30A with bootloader f. I'm not able to use the newest blheli32 FW or anything except 32.1 in this 4th ESC.
New firmware is flashed correctly into this ESC, but motor is not arming. Only when I flash version 32.1 everything is working properly. I think, that it's because of older bootloader version of this one ESC. Is there possibility to upgrade bootloader or fix this issue anyhow? Thanks
[image]

sskaug, 2018-06-20:
Thanks for reporting. We are looking at this with high priority now.

4712, 2018-06-22:
Today was BLHeliSuite32 rev. 32.4.0.7 released, which includes a fix for the issue.
Thanks a lot!

supiiik, 2018-06-23:
Thanks, I will try

supiiik, 2018-06-23:
Can confirm, that it's working now. Thanks :)

4712, 2018-06-23:
Thanks for confirmation and thanks again for bringing this issue to our awareness :)

closed 2018-06-24 by sskaug

simonefpv, 2019-10-31:
hi I also have this problem with esc furling 32 .. I used blheli32suite 32.7.0.3 but nothing doesn't work out