bitdump/BLHeli #315: REQUEST: change increments of Maximum Acceleration

memorystuff, 2018-06-29:
wojld it be possible to change the increments of maximum acceleration from 0.1 to 0.2 or even 0.4? 

Explanation: i fly 6s rigs with stronger higher KV motors i experience sync losses for which i found the only remedy was using max acceleration. everything else failed (demag compensation, higher/lower or auto motor timing, higher/lower startup power etc - tried them all in hours and hours of testing)

i would like to dial in the setting of max acceleration to a point just short of getting these sync losses. with the highest setting around 25%, my wuad flies fine - but i wonder if it would still fly fine with 30% , 40% or even higher. with a bigger increment of the parameter we would be able to set it to higher values - i guess an increment of 0.2 still allows for a very fine grained adjustment.

kind regards,
mickeydee

sskaug, 2018-06-29:
Thanks for the info. I was actually wondering if we should go the other direction, as in some systems (particularly with low rpm protection off) the max accel setting needs to go very low (down in the few percent range). Also, when you set it to 25%, you are hardly limiting acceleration, as you can then go from low power to full power in some 4ms, which is a lot faster than your motors can accelerate (probably they need somewhere between 50-200ms to accelerate).
Did you try setting a current limit to avoid sync loss?

memorystuff, 2018-06-29:
Hi,
thanks for the fast reply. i was also astonished by the results - and the huge effect it made even on 25% 

i did try current limiting - on the bolt32 that is rated for 50a (probably rather a burst rating?) i tried to set it down to about 40a. but did not want to go lower - as what is the point of having a 50a esc, when i limit it by more than 20%. 

maybe we could have max acceleration both ways by introducing a parameter „max acceleration increment factor“ to have the increments either lower or higher than 0.1%. 

for me, the sync losses mostly occured in extreme propwash situations in agressive 180 turns. so it is not so much about accelerating the motors from 0-100%. 

Maybe additionaly worth mentioning: the problem is more pronounced on very new lipos. 

when i flew yesterday with max acceleration at 25 - It felt as if that brought out a little bit more propwash oszilations than without limiting. I could try to do blackbox log to verify that.

kenkamm, 2018-07-13:
Does the Acceleration selling also affect how quickly the motor RPM gets reduced? Or is it only affecting ramp up?

sskaug, 2018-07-14:
The max acceleration only affects increase of power.

kenkamm, 2018-07-22:
Ok. Does the startup power affect braking power?

sskaug, 2018-07-22:
Nope. Braking power is fixed, it can not be altered for a given ESC/motor.

kenkamm, 2018-07-22:
Bummer. Thanks for the reply.