bitdump/BLHeli #317: BLHeli_32 Motor RPM freeze until disarm

FranzPVal, 2018-07-03:
I tested the Beta firmware 32_42 on my new Tekko32 4in1 ESC, but I got a serious issue: After a while one motor (I noticed this for 2 different motors on my quad) keep spinning with a fixed rpm until I disarm. After disarming this motor spin down free running (no brake). Rearm results in continue spinning this rpm. After a power cycle it works again.
I also noticed a problem while I writing parameter with BLHeliSuite32 to the ESC: It stops communicating randomly.

ESC: Tekko32 4in1 (FW: Typhoon32_Multi_32_42.Hex)
FC: Holybro KakuteF4V2
Parameters changed: "sine mode" on, PWM 48kHz
Tested with DShot1200 and DShot600

This ESC works normal with the latest stable 32.4 version. 32.4.3 not tested.
On my other builds with Siskin_ST_Multi 32.4.2 works fine with "sine mode" and 48kHz PWM.

sskaug, 2018-07-03:
Thanks for the info. This seems a strange effect, but it may be related to something that will be fixed in a coming Rev32.5. I'll post this Rev32.5 in a few days, would be great if you could test it then.

FranzPVal, 2018-07-03:
I will test it under the same conditions - thanks for fixing.

JesseDFBlue, 2018-07-05:
Had exactly the same issue when the sine mode was enabled. Happend on an airbot typhoon running dshot1200. After disabling the sine mode quad seemed to fly as expected but did a downgrade a soon as i could.

sskaug, 2018-07-06:
Well hopefully it is fixed - it is hard to say for sure, as I haven't reproduced. But I strongly believe this issue to be gone in the Rev32.5 hex files that are now published.

FranzPVal, 2018-07-06:
Oh no. Sorry but this bug is still there in 34_5. Beside the rpm problem I almost need 2-4 times to write a configuration to the ESC...

How I test: Props off. Quad stay level on the floor. I arm and move a bit roll, pitch, throttle and yaw... After a while a motor keeps the same rpm. Then I disarm and this motor spins down w/o break.

sskaug, 2018-07-07:
I have now tried to reproduce this, without success.
I use the Typhoon32 ESC with default settings, except sine mode and 48kHz pwm.
And Dshot1200 signaling.

What PID rate do you use?
Do you use telemetry?

In my setup I do not have a quad, I only have one FC (Bluejay F4) and one ESC.

FranzPVal, 2018-07-07:
@sskaug Try "sine mode = on". Only with this feature enabled you can reproduce the issue. Btw Do you have also the parameter-write problems?
Both problems don't exist in the last stable 32.4 version.

sskaug, 2018-07-07:
I did have sine mode on.
I have now tried to write settings with all 4 channels connected, and I also see problems. We'll look into this.

sskaug, 2018-07-09:
We have now identified the cause for the parameter write problems, and will post test code shortly. Then we will again try to reproduce the RPM freeze.

sskaug, 2018-07-12:
Rev32.51 test code is now posted. The parameter write problem is fixed. I haven't been able to reproduce the exact rpm freeze as you describe, but I have found potential problems in the code, that are now fixed. Hopefully you can give this code a test :).

FranzPVal, 2018-07-14:
Hi sskaug,

I just tested version 32.51 and 32.52. Both of them seems to be ok - I had no freezes with it. Version 32.51 shows a small bug: current readout was always 0.0A (rpm in osd was ok). With 32.52 the current readout works. In the next days I will fly this in this configuration (sine mode on, 48kHz and motor auto timing). Good work.

sskaug, 2018-07-14:
Thanks for testing! Yes, there was a bug in current (and voltage and temp) reading on 32.51, that's why 32.52 was generated. Let us know if you find any issues.

FranzPVal, 2018-07-16:
I flown 2 packs yesterday with the 32.52 and it seems working well.

My settings (changed): sine mode = on, PWM = 48kHz, motor timing = auto