bitdump/BLHeli #319: BH32 Flashing new MCU

conkerkh, 2018-07-17:
Since I can't get any reply by e-mail looks like no one is working there I will ask here. I have perfectly fine ESC wth dead 4 MCUs. It's pretty much new as it was barely used, since I bought F0 sometime ago I wanna use them to repair this MCU what's the process here? I know KISS who has simmilar authentication mechanism allows for such operation, to me it sounds obvious that's the same case with BH32 since we repair our quads and don't throw them away after they break.

conkerkh, 2018-07-20:
@4712  I'm extremly disapointed by your response. Currently in the quad community it's a common approach we don't throw things away unless they really can't be fixed which obviously you are aware of. It's not really a question of money here but throwing away good things that can be fairly easly fixed with 2 minutes of soldering job. I perfectly understand the desire to protect your IP on BH32 and respect overall work in that got us to where we are at (quads fly great and people are happy). However I'm sure it's possible to find consensus here, so that you can provide service to people who wan't to fix their escs (activate ESCs) I hope you will consider this since you are firmware provider and have authority over authirizing ESCs (apart from different manufacturers).

kimman1, 2018-08-07:
Yep. I have 4 died 32bit esc. I replace it with a new MCU but when i flashing fw on github with my stlink. It's show me a error at 0x00020000
It's very dissapointed if 32bit esc is close source

ju-mpe-r, 2018-08-10:
So, if I understand correctly, there is no way to reflow broken MCU and flash new one on BLHeli boards?

kimman1, 2018-08-10:
@ju-mpe-r Yep

ju-mpe-r, 2018-08-10:
@4712 I have an idea for solution of this situation. Maybe somewhere in BLHeliSuite you can add field for changing old id with a new one(and you will overwrite old id with new one in your database)... So everyone, after buying new ESC can see and save their actual id somewhere and if their ESC died somehow they still will have possibility to fix it.
I think, that you will not loose anything(expect some time), but this can be really cool and usefull feature for BLHeli, and you can anounce and present it. sure community will happy ;)

ju-mpe-r, 2018-08-21:
@sskaug why you are ignoring this thread?

sskaug, 2018-08-21:
It has been responded to by 4712 above.

conkerkh, 2018-08-21:
@sskaug I believe @ju-mpe-r was looking for a reply looking more like this: No, we won't do it, we don't have time to deal with individuals - more direct answer;)

kmitchel, 2018-11-25:
Contact Airbot and ask for mcus.

conkerkh, 2018-11-25:
@kmitchel I did, they were very helpfull. Esc is fixed and flyin for some time.

CreeT6, 2019-04-19:
I killed a mcu on my typhoon32, and I was gona buy some mcu to replace it, I guess i can’t do this with blheli32. Did anyone have success repairing mcu?

mice5, 2019-11-20:
Hi!
I wanted to reuse one of my dead ESCs to use it in simple project.
I've read in BLHeli32 instructions, that UNACTIVATED ESC can operate in 1-2ms range (50-400Hz PMW mode).
Can i flash the .hex file to 0x08000000 and run it "unactivated" ?

closed 2020-07-22 by conkerkh