bitdump/BLHeli #322: Cant Update From 3.4 to 3.5 Missing Record Mark?

AlexSamayoa, 2018-08-19:
Hello,

Downloaded the latest Blheli32 tried to update since the .HEX its not on drop down menu i came to Github for it click save, saved in desk top, opened the New Suite flashed with file but it Terminates the task / command with Missing Record Mark. Closes the Task.?? any ideas i have downloaded again same results for a Siskin_32_Multi ...

Thanks

4712, 2018-08-21:
https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/Rev32.6%20hex%20files/Siskin_ST_Multi_32_6.Hex
You need to right click on "Raw" and "save target as..".
Alternatively download the  whole archive as zip an unpack on your desktop.

rcmodelsgr, 2019-06-06:
any updates? same problem blhelisuite32.6.1.4 trying to flash tmotor_45A_32.64 frmw

4712, 2019-06-06:
@rcmodelsgr goto https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code%20Rev32.6.4%20hex%20files/T_Motor_F45A_4IN1_V2_Multi_32_64.Hex
You need to right click on "Raw" and "save target as..".
Alternatively download the whole archive as zip an unpack on your desktop.