bitdump/BLHeli #324: Feature request - BLHeli_32 Suite for non Windows based PC

TheMrJahn, 2018-08-21:
Hi,

There are a lot of other operating systems besides Windows platform. Why won't create additional versions of BLHeli_32 Suite for MacOS and Linux...

mikeller, 2018-08-21:
@TheMrJahn: The problem is that the BLHeli_32 Suite seems to be written in Delphi, which has little to no portability to anything outside of Windows. So it would need a complete rewrite.

And unfortunately, the maintainers of BLHeli_32 refuse to open source their API, so other projects like @diehertz's BLHeli Configurator are prevented from jumping to the users' aid...

4712, 2018-08-21:
BLHeliSuite32 is indeed written in Delphi, and it is not that easy to do a port, but it hopefully is not a complete rewrite required. There are already steps taken on the way, but there still also remains a lot of work to get all things running on MacOS and Linux.

sskaug, 2018-08-22:
So in addition to MacOS and Linux versions of BLHeliSuite32 being in the works, we are also working on an Android version of a BLHeli_32 tool. Here is what the prototype currently looks like:
[image]

TheMrJahn, 2018-08-22:
@4712 @sskaug Great news, thanks for your work!

r1b4z01d, 2019-02-02:
Great! If you need any testers before beta PM me.

Hect0rG, 2019-02-20:
PM me for beta testing on Linux if needed.

RicBent, 2019-02-23:
I'd love to beta test for Linux as well.

4712, 2019-03-02:
Here a first beta for Linux.
Tested with Ubuntu 18.04.2 LTS.

[BLHeliSuite32xLinux64_Beta_0900.zip](https://github.com/bitdump/BLHeli/files/2922127/BLHeliSuite32xLinux64_Beta_0900.zip)

mikeller, 2019-03-02:
Gave it a quick spin, works out of the box on ubuntu 18.10. Good work!

4712, 2019-03-02:
Wow, you're fast! Thanks!

mikeller, 2019-03-02:
Hehe, been waiting for it for a long time. :stuck_out_tongue_closed_eyes:

RicBent, 2019-03-02:
Thank you @4712! Works great on Manjaro 18.0.3.

skat0, 2019-03-02:
any update for mac's?

RicBent, 2019-03-02:
Actually I found a bug. The music editor seems to break. It worked once and now I always get this:
[image]
and then
[image]

4712, 2019-03-02:
Thanks, I will try to reproduce.

tridge, 2019-03-02:
great to see a Linux version! I tried it on ubuntu 18.04 against a KakuteF4 running ArduPilot. Works nicely for config. When I try to flash an updated fw I get:
[image]
any idea what ssl library I need?

mikeller, 2019-03-02:
@sskaug: Looks like there are some issues with library dependencies not being met. It's probably best in the long term to roll the linux app into `.deb` and `.rpm` installable packages for release, and have these packages declare the dependencies they require (VLC, openSSL). This will allow 95% of users to install the package without having to worry about dependencies.

4712, 2019-03-02:
@tridge: Openssl 1.0.x is the right version. 1.1.x is not supported yet.
@mikeller: I don't know, if packages will be possible, using Delphi FMX.

mikeller, 2019-03-02:
@sskaug: You could always build the executable with Delphi FMX, and then take the files and build the packages in a second step. We use `gulp` to build `.deb` and `.rpm` packages for Betaflight configurator, which might not be the best choice for you as it is geared towards JavaScript builds, but if you decide to go down that route I am happy to help.

4712, 2019-03-03:
@RicBent please try updated version 0.9.0.1. Music editor close exception should be fixed now.

RicBent, 2019-03-03:
@4712 I still get the same errors with the access violation address changed to 0x6AC888.
Just like with 0.9.0.0 the music editor worked for the first ESC but opening it a second time fails. Restarting the app has no effect but power cycling my quad and restarting the app allowes using the editor a single time again.

4712, 2019-03-03:
@RicBent now I got it, I usually edit music for the ESCs all together... but now I found the bug hopefully fixed it in rev 0.9.0.2.
Also fixed an openssl lib load issue.

RicBent, 2019-03-03:
Works perfectly now. Thank you!

TheFrisco, 2019-03-09:
OK so I almost embarassed to ask this but ... Trying to run the BLHeli32 Linux app on my Chromebook, don't own PC, only a Mac and a Chromebook. How do I start the Linux app from the Terminal window in Chromebook Linux VM?

Thanks Robert

mikeller, 2019-03-09:
@TheFrisco: I suspect this won't work - while Chromebooks are running linux, what they are running is an extremely cut down version, with everything that is not needed to run Chrome removed - this most likely means that libraries that are part of every 'normal' linux distribution and are required by the BLHeli32 app aren't present on the Chromebook.

TheFrisco, 2019-03-09:
> @TheFrisco: I suspect this won't work - while Chromebooks are running linux, what they are running is an extremely cut down version, with everything that is not needed to run Chrome removed - this [...]

Thanks for the quick response, and you are probably right but WHAT COMMAND would I even run from the Terminal app to start BLHeliSuite32?

mikeller, 2019-03-09:
@TheFrisco: `<path to wherever you installed it>/BLHeliSuite32xl`

TheFrisco, 2019-03-09:
> @TheFrisco: `<path to wherever you installed it>/BLHeliSuite32xl`

oh gosh that worked, just needed to be outside the folder and voila ... now I need to see if anything connects LOL

THANKS A LOT!!!

TheFrisco, 2019-03-09:
> > @TheFrisco: `<path to wherever you installed it>/BLHeliSuite32xl`
> 
> oh gosh that worked, just needed to be outside the folder and voila ... now I need to see if anything connects LOL
> 
> THANKS A LOT!!!

App runs on ChromeOS with Linux enabled, just can't get the port to show up. Hmmm, open to suggestions. Thanks Robert

mikeller, 2019-03-09:
@TheFrisco: Oh wow, now I am surprised. :grinning:

4712, 2019-03-09:
@ TheFrisco did you try to add user to dialout group?

mikeller, 2019-03-09:
@4712: There is only very limited user management on ChromeOS, and the one user that is used for everything does have access rights to serial ports (i.e. BLHeli Configurator works out of the box), so I think this should not be a problem.

4712, 2019-03-09:
Ok thanks, I will try to setup a VM and see what I can find out.

RicBent, 2019-03-09:
@TheFrisco When connecting your FC you can get access to it via
`sudo chmod +666 /dev/tty####`

TheFrisco, 2019-03-09:
> @TheFrisco did you try to add user to dialout group?

None of that I believe exists in ChromeOS

TheFrisco, 2019-03-09:
Before I try the CHMOD command, I need to vent for a second ... To release product that's only configurable by Windows, that's so 80s :-(

Done venting, happy again ;-)

penkypenk, 2019-03-26:
**@4712:**
First of all I´d like to thank you for your efforts on porting BLHeli32Suite to Linux! Great!
Everything seems to work fine with accessing the ESCs by FC passtrough. There is only one thing which does not seem to work: When I click on the "Motors"-Tab and tick the warning to enable motor control, nothing happens when I ty to ramp up speed on the master regulator. 
When I try to calibrate my ESCs with the button, it signals me an access violation:

[image]

When I click it the second time, i get a different message, that tells me that the component calibrate_ESC already exists:

[image]

Can you checkout that one or is it normal?
Oh and I´m running all on Linux Mint 18.10

Cheers and all the best
Patrick

4712, 2019-03-26:
@penkypenk: Thanks for reporting! I cannot reproduce here with my Mint version 19.1. Could you provide the exact system information, maybe with a link to download the specific iso?
Then, please activate the log function under "Options" and click the warning to enable motor control.
Change to the log tab and do a screenshot.
Thanks

penkypenk, 2019-03-27:
**@4712:**
Thanks for digging in this! First of all I have to excuse myself, because the Mint-Version I gave you, was wrong. I´m actually on Mint 19 (Tara) 64bit. Sorry for the wrong information.
Please find below the image of the log:

[image]

penkypenk, 2019-03-27:
Oh well, just updated my Mint to 19.1 and the error is still existing in the same way.
Maybe it has something to do with the firmware on the flight controller. Currently I´m running Dronin on my copter which does an excellent job so far.

4712, 2019-03-27:
@penkypenk: Thanks, seems to be an issue in combination with dronin.
Which FC board do you use?
What are the errors of MSP_ADVANCED_CONFIG and MSP_SET_MOTOR? (please click on the "+" until the details are shown).

penkypenk, 2019-03-27:
@4712:
Yes thought also that the combination with dronin shows the error. I´ve attached an unfolded log as requested. My FC-board is a Holybro Kakute F4 V2 (not the AIO-Version!) and I´m running a tricopter setup on this...
[image]

4712, 2019-03-27:
@penkypenk: dronin is supporting a sort of msp emulation not including all commands like MSP_SET_MOTOR. Obviously  it also sends "|" and not "!" as error response.
So atm the conclusion is, that you need to do ESC calibration with the dronin gui and I will implement better error handling BLHeliSuite for dronin.

penkypenk, 2019-03-27:
That´s fair enough @4712! Thanks for your efforts...

mikeller, 2019-03-27:
@penkypenk: It's probably a good idea to report this as a bug against dronin as well - especially the fact that they are sending '|' instead of '!' on error sounds much like a typo.

TheMrJahn, 2019-03-27:
Have someone version for Mac?

4712, 2019-03-27:
@TheMrJahn sorry, Mac version is still at work, but it will benefit from the current bug fixes.

tracernz, 2019-03-27:
> @penkypenk: dronin is supporting a sort of msp emulation not including all commands like MSP_SET_MOTOR. Obviously it also sends "|" and not "!" as error response.
So atm the conclusion is, that you need to do ESC calibration with the dronin gui and I will implement better error handling BLHeliSuite for dronin.

Apologies, that is indeed the case: https://github.com/d-ronin/dRonin/commit/9c36c795d7b9198ca413eabe1e7e6dc5ce6cad73#diff-13ccc6598edbdf25cb6e8f9fb12cf078R201

I'll get this fixed up on the dR end, by unfortunately it will take quite a while for that to filter through to end-users. If it's not too much trouble it would be appreciated if you can handle | as error too.

tracernz, 2019-03-27:
Actually, the pipe character is correct for an unknown message: http://www.multiwii.com/forum/viewtopic.php?f=8&t=1516
```
if the message is unknown:
$M|[0][code][checksum]
1 octet '$'
1 octet 'M'
1 octet '|'
1 octet 0
1 octet [unknown code]
1 octet [checksum]
```

4712, 2019-03-27:
@tracernz well, interesting about the pipe char... So one more reason to add it as an 2nd error indication.

tracernz, 2019-03-28:
Yes, but everybody else uses ! so its the de-facto standard. We'll change dRonin to match that in the interests of wider compatibility. :)

vaughnd, 2019-04-02:
No one else using wine to run the BLHeliSuite32 windows executables? I’ve been doing that on Linux for a few months now. Works great.

mikeller, 2019-04-02:
@vaughnd: Yes that is working fine. But having a native version is still nicer. :grin:

vaughnd, 2019-04-02:
@mkeller for sure! Just wondered if people knew there was a temporary solution 🙂

alxsfrm, 2019-04-08:
It worked for me on T-Motor F35A. and Ubuntu 16.4
The problem is when i try to flash local file, without internet connection, i get message "Unable to establish server connection!".
Looks like it always tries to connect to the server, and doesn't allow to choose load local hex file.
I also tried to copy the HEX file to the BLHeli32_HexFiles folder, i get error on load the app, which says "hex File Read Error. Missing RECORD_MARK at file start error."

4712, 2019-04-08:
@alxsfrm: An internet connection is mandatory for flashing, parameter reading and writing is available offline.
Be sure to download the hex file correctly. You might have a look [here](https://www.rcgroups.com/forums/showthread.php?2864611-BLHeli_32-Power-to-perform/page222#post41481041).
In some cases, presumably when saving the file with a MAC, a BOM UTF-8 is falsely added to the start of the hex file. For this issue a fix will be implemented in the next beta release in a few days.

alxsfrm, 2019-04-08:
Why the internet connection is mandatory if you can download the hex file and use it like offline ?
Does it looks for some burning parameters for specific board ?
Is it possible to make the app Flash on offline PC ?

4712, 2019-04-08:
@alxsfrm: An internet connection is mandatory for flashing on all operating systems. Among other things, it verifies if the ESC serial number is registered in the database.
The handling of locally stored files is optional for special purposes,as for flashing beta firmware.

4712, 2019-04-10:
[Updated BLHeliSuite32xl (Linux64) Beta to rev. 0903](https://github.com/bitdump/BLHeli/issues/324#issuecomment-468921159)
Changes in beta 0.9.0.3:
_General:_
- Added 32.6 hex file for:
-- Aikon_AK32_4IN1_55A_6S_V1_0
-- iFlight-35A-BLHeli_32
-- Neuron_40
-- PYRO-32F3
-- ReadyToSky_32
-- Spedix_GS45A_4IN1_V1_1
-- T_Motor_F45A_4IN1_V2
-- T_Motor_F55A_4IN1_PRO_V2
-- XILO_ESC
- Added handling for falsely added BOM UTF-8 in external hex files.
- Added basic workaround for alignment handling under Elementary OS.
- Added basic workaround to improved performance of log function.

_External 4way-Interface (Cleanflight/Betaflight/et al.):_
- Added workaround for Betaflight VCP multiple of 64 byte read failure bug with F7 FC boards.
- Added basic compatibility verification of FC variants with Motor tab functions.
- Added Error handling for msp error sign (“|”) currently used by dRonin.

4712, 2019-04-10:
[image]
Here a first beta for MacOS.
Tested with High Sierra 10.13.6

[BLHeliSuite32xm 0.9.0.3.dmg.zip](https://github.com/bitdump/BLHeli/files/3065350/BLHeliSuite32xm.0.9.0.3.dmg.zip)

[97 more replies]