bitdump/BLHeli #326: Voltage Reading Adjustment - BLHELI_32 ESC Telemetry

RyanSA, 2018-09-11:
Is it possible to implement voltage calibration for BLHELI_32 ESCs in the same manner as with current sense adjustments?

RyanSA, 2018-09-13:
I'm seeing 0.3V deviation using a Typhoon32. I realise that given the process of determining voltage the margin is very small, nevertheless I think it would be something worth implementing.

In terms of current sensing, I'm assuming it is normal to see big differences between actual mah consumed and what is reported?

sskaug, 2018-09-14:
I guess so, current measurement accuracy can vary significantly.

RyanSA, 2018-09-14:
Would it be possible to implement voltage adjustment?

Naterater, 2019-06-02:
This would definitely be useful. I have seen voltage errors near 1v (indicating 26.2v on a 25.2v fully charged battery).

ninja-zx11, 2021-03-26:
Yes that would be great. Mine tekko esc's are .4v off.Please add this feature.

geofrancis, 2021-11-24:
yes, mine are 1.4v off.