bitdump/BLHeli #328: LED contyrol on Holybro Tekko32 F3 35A 3-6s BLHeli32 ESC

utkinpol, 2018-10-12:
Hi, 
after a flash of those ESCs with latest 32.6 code to replace 32.4 code they came from LEDs refused to light up after reboot. 
In the control suite it would light up fine, same as with 32.4 - all was seemingly OK, but after disconnecting power and reconnecting back LEDs would not work. Flashing back to 32.4 resolved the problem.

sskaug, 2018-10-12:
This is strange. Which code does your ESC use?

GeorgeTG, 2018-11-06:
I have strange led issues with the same ESC's since 32.6. I had to disable and re-enable the leds multiple times to fix this, but never managed to get the led from ESC#1 to go red anyway, the rest of the colors work correctly.

utkinpol, 2018-11-06:
Is there any pressing reason to use32.6 instead of 32.4? I am still on 32.4 and it seems to be working fine. Leds were not workingon 32.6 no matter what.

GeorgeTG, 2018-11-06:
Pressing no, I don't think so. I just noticed it's been almost a month and the problem is obviously still there, so I thought I should report it.

sskaug, 2018-11-06:
Sorry for being slow in responding to this. There was a spec change on the timing of this LED from 32.4 to 32.6, I'll check with the mfgr what has happened here. What is the name of the FW that your ESCs use?

vpaulsmith3, 2018-11-07:
hi, i have the same issue with the led and 32.6 when using airbot furling32. is there a fix yet?

Asizon, 2018-11-29:
Furling32 and tekko holybro are the same ESC,and YES,i comfirm the issue with LEDS fail when updating to last blheli32 version

GeorgeTG, 2018-12-02:
What I don't understand is what is the "Code name" or "FW name" that @sskaug mentions. How do we find this out?

Asizon, 2018-12-02:
The firmware name is Furling32,I think that tekko32 F3 use same target @sskaug

GeorgeTG, 2018-12-30:
Just saying this is still a problem. Firmware is Furling32 32.6. Problems are : - sometimes two of the escs won't boot - leds not working - quad feels like a boat in a bad weather

GeorgeTG, 2019-02-04:
ESCs behave oddly even in 32.4, leds still not working reliably, but more importantly sometimes one esc is not responding correctly. Also it's been so many months and there's no update on this..

Asizon, 2019-02-04:
@GeorgeTG but you said that 32.4 fixes everything,did you try 32.62 beta firmware version for Led test?

GeorgeTG, 2019-02-04:
@Asizon at first it seemed like so. But after having enough time (weather got better) to tune and fly the quad, I noticed that some times one ESC wont behave properly. And one morning without having any crashes or anything the leds started acting up again, even on 32.4. And the way the led flickers tells me its a timing issue. Also on 32.4 the telemetry is broken, when it comes to Amps.

mikiha81, 2019-02-11:
having the same issues with furling firmware (32.6)  have the   tekko32 F3 with a matek F722 SE FC 
leds would  go off after  re plugging battery

HyGy, 2019-02-17:
I have the same led issu with furling 32.6. When will be corrected? Thanks.

sskaug, 2019-02-17:
The code is verified with the manufacturer to be correct, so if LEDs do not work properly it is a hardware issue with the unit.

GeorgeTG, 2019-02-17:
@sskaug so all the people claiming there is a problem have faulty units? Seems pretty weird, since **downgrading to 32.4 fixes some issues** (but makes others surface e.g. Faulty current telemetry)

sskaug, 2019-02-17:
Ouch - sorry sorry. I now see that there is an issue with the Rev32.6 codes for "Furling32" and "GetFPV". This is however fixed in the Rev32.6.4 test codes that are already published (here on github).
So if using Rev32.6.4 test code fixes the issue, it is not hardware related.

HyGy, 2019-02-17:
Okay I uploaded the hex file from one ESC successfully (at third attempt), the second I try, failed for the first attempt, then and the second and the third attemt to, then I restarted the copter. And now it is dissappiered, the blheli suite does not see after connecting the pixhawk. So I lost it.

How can I get it back to live????

sskaug, 2019-02-17:
Have you downloaded the latest Suite 32.6.0.8?
Also, pressing Connect first and then Read Setup has been more reliable in some cases.

HyGy, 2019-02-17:
> Have you downloaded the latest Suite 32.6.0.8?
> Also, pressing Connect first and then Read Setup has been more reliable in some cases.

Yes I downloaded, and I started with this. And I always press connect first, then read setup, and then flash. But now the read setup does not found this esc. (and the esc does not beep)

Can I get it somehow back to live?

gvhlaw, 2019-02-21:
I have the same LED problem on a KS 50amp ESC - only one of four fails to show red, green or blue - this started with 32.63 - and continues through 32.65 - both beta hex files for RPM testing.  The LED would only light green after flashing 32.63, now, with 32.65 no lights at all.  Reverting to 32.63 did not allow the light to turn on in green or any other color.  So, three of four ESC'S have working LED'S.  Reverting to an earlier release won't work for use of RPM firmware beta in Betaflight.  Is there a fix?  Is there a way to reset the ESC to its factory configuration?

sskaug, 2019-02-21:
Is this the "KS_BLHeli_32_50A_V1" code?
When you flash the Rev32.6 code, do the LEDs then work as expected?
I am not sure what code your ESC came with, but flashing 32.4 or 32.6 with default settings will be equivalent to going back to factory defaults.

gvhlaw, 2019-02-21:
The answer is that only one color worked out of three when I reflashed to
32.6. I can try reflashing to 32.4 and see if anything changes.  But, then
I can't fly RPM code.

Thank you for your reply.

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

gvhlaw, 2019-02-22:
How many times can you flash an ESC and go back and forth to earlier and later versions before it becomes a brick?

sskaug, 2019-02-22:
An MCU can be reflashed minimum 10000 times. So that is not likely to become a problem...
I'm surprised you have LED problems with this "KS_BLHeli_32_50A_V1" code ESC. It only uses gpios for LED control - no serial LED with potential timing issues. Maybe your ESC has problems?

ITSFPV, 2019-02-22:
I notice that the number of bad frames increases the more you flash an ESC, is that to be expected and at which point does become a problem?

HyGy, 2019-02-22:
How can I directly flash the MCU if I lost the communication trough this 1 wire protocol?

carloscampos775, 2019-02-24:
LED control not working for me 32.6.5. LEDs will also randomly light up. 
[image]

Asizon, 2019-02-24:
@carloscampos775 your esc are a metal wraith, we are talking about the Furling that are F3, and I see that the firmware problem persists, 16.4 let activate LED but not change color, the last .6 directly do not let activate esc. Why not open the blheli code so that more people can develop?

gvhlaw, 2019-02-26:
I agree!  Or please move faster to correct these issues.  I now am in a
pickle after flashing the last Beta hex - I lost my USB plug, and I have
one ESC that is behaving strangely and with no LED.

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

sskaug, 2019-02-26:
Now some issues with serial LED handling have been identified and fixed. Please try the Rev32.6.6 test code that is posted here on github and in the Android app.
Please note that bootloader rev j has some limitations for "GetFPV" and "Furling32" codes:
- LEDs will not change when pressing Write Setup. You will have to press Disconnect or power cycle the ESC for LED updates to take place.

HyGy, 2019-02-26:
How can I reflash the ESC if I lost communication?

sskaug, 2019-02-26:
Unfortunately if you can not communicate with the ESC, it can not be reflashed.

HyGy, 2019-02-26:
> Unfortunately if you can not communicate with the ESC, it can not be reflashed.

Not with an external HW? Like an arduino or stg? This is just a MCU, all the others can be flashed somehow.

gvhlaw, 2019-02-26:
Thank you for fixing this - as soon as I have a new FC with a USB port, I
will give it a try!

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

mikeller, 2019-02-27:
@HyGy: As far as I am aware, the MCU is locked down to prevent ~users~ thieves from stealing the valuable IP that is on it.

HyGy, 2019-02-27:
> @HyGy: As far as I am aware, the MCU is locked down to prevent users thieves from stealing the valuable IP that is on it.

That is crazy. I bought a HW to use, wanted to ugrade the firmware, meanwhile it is unsucessfull somehow, and now I cannot use the HW anymore? The developers need to find out a solution!!!

gamaxgbg, 2019-03-04:
I also have 2 tekko32 (not f3) which i lost communication. They were perfectly ok, then i think they bricked when i tried to flash the FC (this is the only possible reason, because they were ok and after some few FC reflashing they suddenly stopped booting) so i tried to connect to blhelisuite and they would not even read as corrupt, the blheli32 suite simply showed some dashes.

Now i’m going to buy an arduino to try to flash it, but i don’t even know how to do that, because in the esc there are pins that i don’t know what they do: SWC and SWD (and also gnd and one i can’t read exatcly but i think it says 3V).

Is there a way to recover communication? I do not want to buy an arduino for nothing.

sskaug, 2019-03-04:
Bricked ESC after FC flashing sounds very unlikely. Have you tried swapping ESCs on the FC outputs to see if the failures follow physical ESC or FC channel?

gamaxgbg, 2019-03-04:
Sorry, actually the pin on the fc broke (don’t know how, because it was ok and after some minutes was broken, and the fc and this esc were brand new). 

But the other esc does not even beep when i turn it on, and it was ok when it was in my old quad and don’t know how it stopped working. The leds blink when i plug the lipo and then stop, just as any other tekko32 i have. I have tried to flash it via an arduino nano, but could not connect to it.

tgreer, 2019-03-30:
I have several of these ESCs, which don't light up. I tried via the FC. I also made a 4 way interface and tried that way. I've put them on 32.68 beta. 2 of them will light up. 2 wont. Are we certain it's fixed in this release before I go back to my supplier?

sskaug, 2019-03-31:
If your LEDs do not work with 32.6.8, I'd say you should contact your supplier.

tgreer, 2019-03-31:
Understood, I will do that thanks

Regards

Thomas

tgreer, 2019-05-20:
Having replaced all the ESCs, I believe this is still an issue in the code. I set the LED colour, reboot, and sometimes it stays set, othertimes its no longer set and wont come on anymore. Sometimes they randomly light.

MAXimumEFFORT843333, 2019-05-30:
very new to Github but all I found for codes is the 32.62 and one of the 4 single escs LEDS are still not turning on.... and this is the 3rd esc that has been replaced to see if it was the esc and not the firmware...

VooPoc, 2019-06-05:
I've recently purchased the Holybro Tekko32 F3 35a.

I turn my ESC LED's off and sometimes various ESCs LEDs start working when I connect a battery.  I am on 32.6.  

Additionally and more important IMO, some ESCs sometimes don't arm.  I need to reconnect the battery.  These are new ESC's and are doing this out of the box.  This was originally one ESC and I swapped its position to troubleshoot.  This issue followed the ESC, however after this two ESC's started not arming.

I also have a 1000uf capacitor connected to the battery terminal (not sure if this is affecting start timings), PWM freq. is 48 kHz and using dshot1200.  For my maiden flight, I left it stock (24 kHz) on dshot600 and for 2 packs and lots of disconnecting and reconnecting, I did not get these issues.  What are peoples configurations?

ju-mpe-r, 2019-06-28:
I have only two of four esc's with working led
TEKKO32 F3 METAL with firmware 32.6.8
Any suggestions?

ju-mpe-r, 2019-06-29:
@sskaug ?

sskaug, 2019-06-29:
I have brought this issue up with Airbot. Together we may find out what is going on.

ju-mpe-r, 2019-06-29:
> I have brought this issue up with Airbot. Together we may find out what is going on.

Can I help somehow?

sskaug, 2019-07-04:
Airbot have now checked a batch of their outgoing products and found no issues. So your units may be defective and subject to return and refund.

GeorgeTG, 2019-07-04:
Can confirm my ESC was probably from a bad batch. Everything was fixed when it was replaced. Now running latest beta with bidirectional dshot and RPM filters and everything works fine 👌

MAXimumEFFORT843333, 2019-07-04:
Can someone put a link to the updated firmware I'm unable to,find it. Also if airbot says that previous batches may be defects have they told their supplies to replace said defects?

tgreer, 2019-07-04:
Hmm. I’ve replaced the full batch and had similar problems. Will give them another go.

Regards

Thomas

ju-mpe-r, 2019-07-06:
What? Dead RGB's? Really?

ju-mpe-r, 2019-07-07:
Only two escs(of four) with working LEDs on TEKKO32 F3 METAL with firmware 32.6.5 also.
Do I need test LEDs separately?

VooPoc, 2019-07-07:
@sskaug do you know there testing strategy, I'm on 32.6.4 and over 6 packs I usually get one time some ESC's don't start and show this LED issue (I turn LED's off and they turn themselves on).  I've also noticed some differences if I change some settings, such as protocol and PWN freq.

ju-mpe-r, 2019-07-08:
Also have the same situation like @VooPoc with one ESC won't start once of 4 packs.

ju-mpe-r, 2019-07-29:
@sskaug any news on LEDs issue?

[7 more replies]