bitdump/BLHeli #330: Observation - Change Hardware Definition

louis-test, 2018-10-18:
Background: I had an MCU (efm8bb2) fail on a 4-in-1 due to a crash. In an effort to repair this ESC, I pulled another efm8bb2 from a donor ESC and replaced the failed one. These are different model ESC with different layout file (J-H-25 vs. A-H-25).

Observation: After jumping the signal wire to pin 4 on the newly donated mcu - I can access the device through blheli-configurator (chrome). I attempt to update the device to A-H-25 through the configurator and get 'success' messages - however the RC pin remains pin 4 (not the expected pin 5 as in A-H-25).

Only when flashing the donor MCU through the C2D/C2CK interface am I able to actually get A-H-25 loaded onto the MCU and get the RC pin to move to pin 5.

Comment: Could you adjust the flash read/verify to include the bootloader code? I was mostly thrown off by the fact that both blheli suite and blheli-configurator both confirmed that A-H-25 was loaded and okay even though the pinout clearly reflected J-H-25.

Thanks,
 Louis

DroneMesh, 2019-04-19:
How did you figure the pinout.. I am about to release open hardware blheli_s esc.. Before i release to public I want to make sure my sense pins and the gate pins are matching..

Can you just clarify where to read the pinout.

Thanks

louis-test, 2019-06-24:
Hi DroneMesh, sorry I missed your response. I had a badly burned 4-in-1 ESC that I pulled the parts off of and carefully traced out. I used the code here to double-check my measurements / observations. You can figure out the 8051 pinout from just the *.inc file but there isn't any solid documentation on the gate driver.

I'm happy to share the schematics I created with you if you're interested. Blheli_s ESC are very simple.

emin0299, 2024-07-09:
Hi, How can I figure out connection pin? I'm designing an esc based on atmega328. Also I use bemf methode and opamp comparator (lm339) for zero cross sensing. I've confused a little bit to which pin connect to which pin? For instance, opamp output or gate drivers. Because I want to use blheli firmware however I didn't find any information guide.