bitdump/BLHeli #333: RS485

fvantienen, 2019-01-14:
Could there be an option to add a digital protocol like RS485 to the BLHeli_32 firmware?
We are interested in having a fully digital communication protocol for our ESC's and are interested in BLHeli ESC's.

sskaug, 2019-01-15:
At the moment BLHeli_32 supports what the RC community receivers and flight controllers use. And as this is currently by far the dominant application area, applicable input signals is governed by these.

fvantienen, 2019-01-15:
We are willing to create our own hardware, but since BLHeli_32 is closed source we can not create the correct firmware for it. That is why I asked here if there is a possibility to add this feature to the firmware for us and work together on this?