bitdump/BLHeli #336: ESC Telemetry doesn't work after enabling signal line RPM telemetry

SwankyPantie, 2019-01-29:
I've installed the latest 32.62 test version to try out signal line RPM telemetry. That part is working fine although telemetry coming from the telemetry line is all messed up. Connecting the battery reads 13-15A prior to arming and pretty much all the other telemetries are skewed and jumping around. 

Disabling Signal line RPM telemetry fixes the problem, in fact telemetry line is stable until the FC sends the DSHOT command to enable signal line telemetry

I'm using Holybro Tekko32 F3 ESCs

sskaug, 2019-01-29:
So I've found the issue. In order to save DMA BW, the ADC DMA is turned off while doing the signal line DMA. And in this disabling/enabling of the ADC, the sequency of the ADC samples is screwed up (for the GD32F350, not for the STM32F051 - their ADCs are different).
I'll fix this - will be a couple of weeks before I have time

sskaug, 2019-02-09:
This issue is now understood and a fix is implemented. But before releasing code I'd like to understand what happens in issue #337.

sskaug, 2019-02-10:
New test code Rev32.6.3 is now uploaded on github and available in the Android app.

SwankyPantie, 2019-02-10:
@sskaug I've flashed my ESCs to this version. Current measurement stopped working altogether, it reads 0A. When connecting the battery I see a momentary surge reported but after that no readings at all.

sskaug, 2019-02-11:
I can not reproduce this here. I use Rev32.63 Furling32 code with rev h bootloader.
I get good current readings in BF:
[image]

SwankyPantie, 2019-02-12:
Rough motors come from the timing difference between Gyro and PID loop. When enabling dshot_bidir BF enables RPM filter which requires the Gyro and PID sampling to be the same.

Here's what I see: 
[image]

This screenshot is with a fresh firmware, everything is default and motors are not running. Both Current and Voltage metering is off.

SwankyPantie, 2019-02-12:
For our records, here's the same screenshot with the same firmware and settings as above flashed back to 32.6

[image]

sskaug, 2019-02-12:
Tried another ESC now, and I can reproduce a similar issue. I'll dig into it.

sskaug, 2019-02-12:
@SwankyPantie 
Ok, so I believe I have found the root cause.
I have published a 32.6.4 version of the Furling32 code on github.
I see no problems in my testing here. Could you please test it?

SwankyPantie, 2019-02-13:
@sskaug I've tested 32.6.4 and ESC telemetry seems to be working again.

[image]

It is not visible on this screenshot but Voltage telemetry works as well. Thanks for fixing it!

sskaug, 2019-02-13:
Great. Thank you for reporting and testing!

closed 2019-02-13 by sskaug