bitdump/BLHeli #337: BLHeli32.62 - BLHeli suite not always be able to connect to ESCs

SwankyPantie, 2019-01-30:
After flashing the test code, BLHeli Suite is not always be able to connect to the ESCs. It fails to read the configurations after connecting and I'm only able to read the configurations one by one. This works for ESC 1,3,4 after a few tries. For ESC 2 I'm not able to read the configuration at all, thus at this point I don't know how can I flash it again 🤔 

I've flashed back 32.6 for those 3 ESCS and the Suite is always able to connect and read config for the first time.

ESCs Holybro Tekko32 F3 (Furling Multi)

sskaug, 2019-01-30:
This may be related to your other issue (#336). The ADC DMA must be disabled during bootloader operation, and this is likely not done properly for GD32F350 MCUs.

SwankyPantie, 2019-01-31:
That makes sense, actually... I'll try to connect to the ESC using arduino, worst case I hope I can connect directly to the MCU and flash it forcefully if nothing works :/

sskaug, 2019-02-09:
What is the exact name of the firmware in your ESC?
And which revision of te bootloader does it have?

I've tried to reproduce this here, with 2 psc F3 4in1 ESCs that I have. And the Suite sometimes has problems detecting all 4 ESCs. For both 32.6 code and 32.6.2 code. But I have always been able to connect when selecting only a single ESC. 

Using the BLHeli_32 app, I can always detect all 4 ESCs, but then again it fails during flashing (from 32.6). I'll dig into what happens with the app.

Have you been able to flash your last ESC back to 32.6?

To me this seems like issues with the Suite/app, maybe related to timing.

SwankyPantie, 2019-02-09:
The full name reported by the suite is **Furling32 - Rev 32.62 - Multi** and it says **BlHeli32 Bootloader h**  (i don't know if "h" is the version or not...).

In my case, the symptoms are similar too. After connecting and reading all configurations, it always fails. Clicking on individual ESCs and trying to read them one by one works for ESC 1,3,4 but even single read doesn't work for ESC 2.

I can flash those 3 ESCs back and forth but not ESC 2. I haven't tried to solder out the other 3 ESCs and connect this one to Motor 1 pad. I'm going to try that to recover it once there's a solution available or doing it via Arduino, idk.

I don't have an Android phone at hand, thus can not test it with the app.

4712, 2019-02-09:
Could you try to "Connect" in BLHeliSuite32 first, then the power up the ESCs and after 2 or more seconds "Read Setup" ?

SwankyPantie, 2019-02-09:
@4712 wow, that did the trick :O I was able to connect to **all 4 ESCs** and read their configs! Thank you

4712, 2019-02-14:
@SwankyPantie  Please check with the latest revision 32.6.0.8.
Hope it's fixed now.

ITSFPV, 2019-02-14:
The payload for 32.6.0.8  on both listed sites is infected with the Win32.WGeneric.xlukg  virus.
Unique Threat ID: 223329000

Please clean

skorzo, 2019-04-14:
> After flashing the test code, BLHeli Suite is not always be able to connect to the ESCs. It fails to read the configurations after connecting and I'm only able to read the configurations one by one. [...]

Similar problem on Furling32 ESC  (standalone version) but running blheli32 32.4 and blhelisuite 32.6.1.1

agrath, 2021-08-23:
@4712 +1 for your suggestion. I'm trying to get some old SN16A ESCs running BLHeli 14.9
I could not get one of the escs to come up after flashing but following the steps you gave, it comes up every time. Weird stuff but you just saved me a full replacement stack!

Bartzacula, 2022-06-16:
> Could you try to "Connect" in BLHeliSuite32 first, then power up the ESCs and after 2 or more seconds "Read Setup" ?

That's work for me...Thank you!!!    

Best regards from Chile