bitdump/BLHeli #338: BLHeli32.63 - T_Motor_F55A_4IN1_PRO_Multi_32_63.Hex smoked a motor

ITSFPV, 2019-02-10:
Just flashed my  T_Motor_F55A_4IN1_PRO_Multi_32_63 from T_Motor_F55A_4IN1_PRO_Multi_32_62

On going to motors tab post reboot to check motors, initially nothing happend.  I re-flashed and repeated the procedure.  This time,  one motor caught fire.

Motors\ESC was fine on 32_62 firmware.

sskaug, 2019-02-11:
Ouch - toasted motor is not good.
I have repeatedly done what you describe, using the T_Motor_F55A_4IN1_PRO firmware, but have seen any erratic behavior or problems.

ITSFPV, 2019-02-11:
it was just moving from 32.62 to .63

sskaug, 2019-02-12:
I also tested that several times

ITSFPV, 2019-02-12:
it smoked, when using BLHeli motor test functionality, post rebooting of ESCs & reconnecting of power.