bitdump/BLHeli #339: Own ESC design implementation not working

Nasshh, 2019-02-13:
Hello,
    I am designing own ESC using open source BLHeli Firmware. I have flashed (mystery 12A hex file) .hex file into ATmega8a (32 pin) microcontroller using USBASP and the program is successfully flashed to controller. I have connected the circuit on the breadboard as per the attached image link below, but motors are not running. Kindly help.

Image Link: [url]https://drive.google.com/open?id=1mmfWA1mD_dkv2lf5yPRe2Y0LvSqndF2p[/url]

Components Details:
N-MOSFET                IRF540N
P-MOSFET                 IRF9540N
CONTROLLER          ATmega8a (32 pins)
CRYSTAL                      16MHZ
RESISTORS                  470R, 220R, 100R, 3.3k, 10k, 18k, 4.7k
CAPACITORS             0.1uF, 22PF
TRANSISTOR              PN2222A
LIPO BATTERY           7.4V
PWM GENERATOR   +5V
BLDC MOTOR             EMax PM1306-3000kv

Thank You

[image]

louis-test, 2019-06-24:
Couple comments. 

1.) Your gate resistors are overly large (100ohm).  That, combined with the fact you have a driver on the highside means you might run into cross-conduction issues. Make sure you pick a hex file with the highest possible deadtime and measure this.

2.) The NFET you have on the bottom really isn't designed for 3.3v Vgs and you will very likely have turn-on issues at current. When looking for a FET try searching for they keyword "logic level" and check the Vgs vs. Rds(on) curve. If you don't want to muck about with that - you can also use a gate driver and switch to having the same NFET for both highside and lowside.

Let us know if you get it working well.

thestafman, 2020-06-11:
I don't think the 100ohm is the issue, it's the 470ohm. The 100 ohm doesnt even need to be there. Any luck since.