bitdump/BLHeli #340: help where/ who sell chip stm32 to repair esc  ?

jonathan2018, 2019-02-14:
hi Developer 
i and  my friends had so much esc blheli 32 4in1 or singe esc 32 burn chip STM32. So we want to  repair it instead buy new . 
we knew blheli 32 close source and  we happy to pay money for  that . we tried to contact some  where product esc but they do not answer or do not sell .
or any idea better to repair esc 32 bit ? Thanks

SwankyPantie, 2019-02-14:
imho the MCU rarely dies in an ESC, most of the time the FET burns out. Are you sure it's the MCU what's dead in those ESCs? you can replace the FETs without any issue (assuming you use the same type).

jonathan2018, 2019-02-14:
sure stm32 dead. Fet no problem .i measure them . Problem if buy new chip from dikey  also can not use for firmware closed

kmitchel, 2019-02-18:
Try airbot.

BlurChaser, 2019-03-13:
@kmitchel i am also sitting here with a 4in1, which has a dead MCU...
Can you tell me how or where to contact airbot (i never heard for airbot) to get one mcu and get flying again?
I really dont want to spend a lot of money, if this could be a simple reflow...
Thanks!!

kmitchel, 2019-03-13:
mail@myairbot.com

BlurChaser, 2019-03-14:
Omnibus right, i feel stupid now... 
Ok i will try to contact, many thanks!

RicoKirstein, 2019-04-25:
@kmitchel @BlurChaser Did you have any luck getting new MCUs from airbot? I also have a Ori32 with a shorted out MCU but no answer from airbot yet. If not, I have 3 Ori 32 mcus to give away...

BlurChaser, 2019-04-26:
@RicoKirstein Yes they contacted me back, but it took a while. they were very kind, explaining that they cant send me an mcu, because i have a dalrc esc, and they only have their (airbot) bootloader, so that wont work on my esc... Then i contacted dalrc on multiple emails, but they just ignored me, so i bought a new esc...

you need to be patient, maybe leave a mesage on their contact page. 
good luck with fixing the board!

SolarWang, 2019-06-10:
www.taobao.com/ might help

mikeller, 2019-06-10:
@SolarWang: Not sure you understand. The BLHeli32 firmware is locked to the id of the MCU - replacing the MCU with one that does not have an authorised id will mean it is impossible to flash firmware onto it, so the ESC will not be usable again with BLHeli32.

RicoKirstein, 2019-06-10:
exactly. I also did not receive any answer from airbot. It kinda makes me want to move away from 4in1s or blheli32... sadly there is no real alternative

ju-mpe-r, 2019-07-08:
You can try buy cheapest blheli32 ESCs on bangood and get MCUs from there to your ESC.

Johnson117, 2020-05-15:
> @RicoKirstein Yes they contacted me back, but it took a while. they were very kind, explaining that they cant send me an mcu, because i have a dalrc esc, and they only have their (airbot) [...]

I have an esc(AIRBOT Furling 32 45A) and sure the mcu is dead.You have contacted Airbot.If I replace the MCUs myself, can they give me the firmware?

RicoKirstein, 2020-05-15:
I was very patient. Waited ~4 month. They answered that they could send me a new mcu. Send them my details and wanted to pay with paypal, never heard from them again.