bitdump/BLHeli #343: Found 2 ESC with simular Unique ID numbers

ITSFPV, 2019-03-03:
I get the following error message when trying to read\check the ESCs.   Error: Found 2 ESC with Simular Unique ID numbers,  after which it reboots the Flight Controller.

This is a brand new Airbot Typhoon32 v2 4in1 ESC 4x35A  with the vendors default firmware on. 

I have tried apply power before USB and visa versa but with the same outcome.

4712, 2019-03-03:
@ITSFPV Could you please tell me the BLHeliSuite version version number?

ITSFPV, 2019-03-03:
latest  32.6.0.8

4712, 2019-03-03:
Most likely a fault somewhere, which lets connect one ESC at two different channels. How does it behave when you test motor spin functionality?

ITSFPV, 2019-03-03:
ok very weird.
Motor Tab       spinning motor
1                             2
2                             3
3                             1
4                             3
I have Joe Lucid F7 Bidir build on the FC.  Will flash back to master to see if makes a difference

ITSFPV, 2019-03-03:
flashing back to master (BF4) made no difference

mikeller, 2019-03-04:
@ITSFPV: What flight controller are you using?

ITSFPV, 2019-03-04:
Kakute F7

mikeller, 2019-03-04:
@ITSFPV: Try again with build 1487 when it's done: https://ci.betaflight.tech/job/Betaflight/1487/

ITSFPV, 2019-03-04:
same outcome

mikeller, 2019-03-04:
Looks like 4 way ESC forwarding is broken at the moment in development versions. Use a released version of Betaflight to flash / configure ESCs.

4712, 2019-03-04:
Tested here BF4.0.0#1487 with OmnibusF7. Works fine...

mikeller, 2019-03-04:
@4712: I suspect it's dependent on the pins that an individual flight controller uses for motor outputs. Will have to go over the 4 way ESC code to make sure it works with dynamic resource assignment before 4.0 is released.

4712, 2019-03-04:
@mikeller perhaps, but if you look at the [results of the motor test](https://github.com/bitdump/BLHeli/issues/343#issuecomment-469069147). it might not not have anything to do with the 4way-if implementation.
@ITSFPV did you try to test the motor spin up also with the Betaflight configurator after fresh booted FC?

ITSFPV, 2019-03-04:
so it its the F7,  I just plugged in the esc to a kakute F4v2  and all motors act as expected and BLHeli32 suite can connect and read each esc. Unfortunately, I do not have another kakute F7 to be able to say if its hw specific issue or not.

4712, 2019-03-04:
@ITSFPV: did you try to test the motor spin up with Kakute F7 also with the Betaflight configurator after fresh booted FC?

ITSFPV, 2019-03-04:
@4712  Same behaviour via both means

4712, 2019-03-04:
@ITSFPV I would suggest to flash BF 3.56 to check if it is an issue of BF 4.0 or the hardware.

ITSFPV, 2019-03-04:
ok, looks like hw then as it does it on 3.5.6. sorry for troubling you.  Damn that was an expensive failed firmware upgrade

closed 2019-03-04 by ITSFPV