bitdump/BLHeli #344: Bidirectional Dshot not working on BLHeli_32 Rev 32.6.7 together with F7

WinniNhm, 2019-03-08:
Hello,
I was successfully running bidirectional Dshot and RPM filter on Betaflight 4.0 #1490 and Joe Lucid version from https://github.com/joelucid/rpmfilterf7 together with
BLHeli_32 Rev 32.6.6 on an Aikon_AK32_4IN1_35A_6S_V1_0 / Omnibus F7V2 combination on 8K loop rate and Dshot1200.
Updating to 32.6.7 is causing Motors 1 & 2 to stop spinning after some seconds. The motors are kept “braking” at this point.
I have just seen Joshua Bardwell's video https://www.youtube.com/watch?v=vz4Fo3nftW4
He got stuck with the same BLHeli Re 32.6.7 version.

joelucid, 2019-03-08:
@sskaug, this is interesting: a reported regression in 32.6.7 with 8k/8k on f7. @WinniNhm, 4k worked, correct?

WinniNhm, 2019-03-08:
Just testing with BF 4.0.0 RC1 8k Dshot1200
BLHeli_32 Rev 32.6.6: works with 0.1% - 0.2% invalid packets
BLHeli_32 Rev 32.6.7: fails after about 10 to 60 seconds with almost 10% invalid packets

WinniNhm, 2019-03-08:
BF 4.0.0 RC1 4k/4k Dshot1200 with BLHeli_32 Rev 32.6.7: working nice with hardly any invalid Dshot packets. The issue is with 8k timing.

jtroutt19, 2019-03-09:
BF 4.0.0 RC1 8k/8k DSHOT 1200 With BLHELI_32 REV 32.6.7: Was working for me but I was having issue with bad oscillations at hover. Invalid Packets was extremely high. I flash BLHELI_32 Test code REV 32.6.4 and invalid packets are still slowly increasing but the oscillation is gone. 

I am using a Kakute F7 and a TEKKO 32 4in1 35a.

sskaug, 2019-03-09:
@joelucid There are some other small changes from .6 to .7, but as all is well on 4k and the problems start on 8k, I'd say the problem is likely to come from the additional 5us response delay. In order to push the ESC side, I ran an ESC on the bench at 8k with Dshot1200/600/300. And the ESC still responded well, even with Dshot300 where the packets overlapped. Are you able to reproduce the reported issue?

WinniNhm, 2019-03-10:
@sskaug and @joelucid 
This issue got resolved with the new timer and dma settings for the OMNIBUSF7V2 FC provided by Joe. 8k/8K/Dshot1200 works with 32.6.7 with very low invalid packets rate.

Thank you again, Joe

# status
MCU F745 Clock=216MHz, Vref=3.30V, Core temp=77degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3068, Max available config: 32768
GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
System Uptime: 681 seconds, Current Time: 2019-03-10T13:27:51.895+00:00
CPU:22%, cycle time: 126, GYRO rate: 7936, RX rate: 150, System rate: 9
Voltage: 2243 * 0.01V (6S battery - OK)
I2C Errors: 0
SD card: Manufacturer 0x3, 31166976kB, 03/2018, v8.0, 'SC32G'
Filesystem: Ready
Arming disable flags: CLI MSP
Dshot reads: 20629780
Dshot invalid pkts: 652
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 14 7 7 7 14 7 7 14 7 7 7 14 7 0 0 7 13 7 13 7 13 7 0 7 13 7 0 7 13 
0 7 0 7 0 13 0 13 0 6 6 6 13 6 13 6

closed 2019-03-10 by WinniNhm

RubberDuck00, 2019-03-13:
I have similar issue with KAKUTEF7 4.0.0RC1 / Holybro metal 4in1 on 32.6.7.  Motor Dshot RPM Motor count is zero after booting with battery then plugging into PC.  Motors spin then check status.  I've tried 4k/4k at DSHOT600 and DSHOT1200, same results. I'm using default rpm filter settings as there aren't any specific to this FC.   Ideas?

# status
MCU F745 Clock=216MHz, Vref=3.31V, Core temp=65degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3061, Max available config: 32768
GYRO=ICM20689, ACC=ICM20689, BARO=BMP280
System Uptime: 27 seconds, Current Time: 2019-03-13T05:37:44.836+00:00
CPU:9%, cycle time: 251, GYRO rate: 3984, RX rate: 33, System rate: 10
Voltage: 1490 * 0.01V (4S battery - OK)
I2C Errors: 4
SD card: None inserted
Arming disable flags: RXLOSS CLI MSP
Dshot reads: 327276
Dshot invalid pkts: 4
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 0 0 5 12 7 14 8 16 10 18 12 19 13 1 15 3 
0 0 0 0 0 0 0 0 0 7 7 8 8 7 8 8