bitdump/BLHeli #347: RPM filter - Holybro Tekko32 35A -  Airbot_Wraith32_ST_Multi_32_67 and 32_65

Testmangh, 2019-03-14:
Following the steps https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

With the firmware Airbot_Wraith32_ST_Multi_32_67
In the betaflight motor tab if I set the motor to 1100 (with page up) it runs nicely but if I rise its speed with the arrow button it stops running and the ESC makes the fourth tone, when I zero its value the ESC makes the fifth tone (tested with kakute F4 V2 and kakute F7). 
https://youtu.be/vz4Fo3nftW4?t=1410

With Airbot_Wraith32_ST_Multi_32_65.Hex
The motors spins normally when I rise its speed with the arrow button but the Dshot invalid pks are consistently 1,43%

sskaug, 2019-03-15:
32.6.7 has a fix to the stall protection that makes the ESCs halt after a few start attempts. Typically when you raise the throttle a little, the motor spins a few initial rounds on a boosted startup throttle, then it can stop again. And in 32.6.7 if this happens a few times then the ESC will start to detect input signal again (this is what you describe as the fourth tone). This process upsets bidirectional dshot.
So currently 32.6.7 has issues upon detected motor stalls.
A 32.6.8 will be out soon where this is fixed.

Testmangh, 2019-03-15:
Ok, thanks for explanation =)

BRadFPV, 2019-03-17:
Had a motor stall today on BF RC2 RPM on DYSF4PRO target and Airbot_Wraith32_Plus_ST_Multi_32_67. I'd be happy to test 32.6.8. Please email hex to brad )at( B-RadFPV dotcom.

gorbach, 2019-03-25:
Should I use **Airbot_Wraith32_ST_Multi** target for **Holybro Tekko32 4in1 esc (35A)** esc ? Did I understand correctly ?

sskaug, 2019-03-25:
You must use exactly the same code as the ESC currently has (which shows up in BLHeliSuite32 when you connect to the ESC).