bitdump/BLHeli #348: Minimum PWM Frequency with OneShot Protocol

dusan19, 2019-03-15:
Is there a minimum PWM frequency limit when running oneshot? By generating oneshot PWM pulses with different frequencies we found out that the ESC responds (recognizes the input signal) for frequencies higher than 270Hz. Unfortunately we run our control logic on 250Hz so the ESC doesnt recognize the input signals. We do NOT hear high-low beeps that correspond to signal detection, when the flight controller sends 250Hz oneshot PWM singals.
When we set 1-2ms PWM protocol it works fine with 250Hz PWM signals.
Any help is much appreciated. Thanks

sskaug, 2019-03-16:
Yes, this is correct. When detecting input signals, BLHeli_32 looks for 10 valid pulses/frames within a given time. And for Onshot125, this is 30ms (for 1-2ms pwm, it is 250ms). I can increase the time for Oneshot125 to 100ms, and include this in a Rev32.6.8 test code that I will publish shortly.

dusan19, 2019-03-18:
Thank you for the response. Shouldn't the minimum frequency then be 333Hz? Since each of the 10 frames can be max 3ms?
If you could increase this time, it would be much appreciated, so slower control loops can also use this protocol.

closed 2019-03-26 by dusan19