bitdump/BLHeli #350: Emax mini 2 Magnum / Emax Formula 45A Multi v32.67

JuStAdsor, 2019-03-21:
Hi!

I have flashed Emax mini magnum tower which contains Emax Formula 45A Multi ESC with the 32.67 version in order to enable RPM, but the following issues arise after flash of all 4 ESC:
1. Motor 1 started spinning by himself without being armed.
2. Motor 2,3,4 were not controllable at all, meaning that they could not even be armed.
3. Blheli_32 software was unable to recognize all the ESC; only one was recognized
4. after multiple retries I was able to finally get a connection to all 4 (but marked as unknown) and flashed them right away to a stable version: 32.6. The motors started to work again.

Any solution to this?
Thank you!

sskaug, 2019-03-21:
We are aware that there are issues with the 32.6.7 code. One being a conflict of pullup/down between the FC and the ESCs, that can cause motor to spin without being armed, and also problems in connecting to the ESCs from BLHeliSuite32.
If first powering the FC (via USB), and then powering the ESCs, the connection between ESC and BLHeliSuite32 is reliable.
We are working on fixing this, and expect to have a 32.6.8 out shortly.

JuStAdsor, 2019-03-21:
I see. thank you for the answer. I will try again when 32.6.8 comes out and I will let you know.

Thank you!!