bitdump/BLHeli #353: motor went full throttle after ESC initialized

ExaflopS, 2019-03-27:
After flashing and testing new beta fw blheli 32.6.8 on quad with escs holybro tekko 32 - 35A standalone - so using firmware Airbot_Wraith32_ST_Multi_32_68.Hex , testing and flying with BF4 with RPM filters and bidir dshot enablet - using dshot 1200. 
After connecting 20 th battery pack immediately after esc initialization tones, motor 3 went full throttle without me even touching radio- so it was disarmed. After quad was flipped, motor 3 had bent prop which was knocking on the arm like trying to spin up, after restarting all escs and quad was working fine like before. Before I was testing on the same setup with blheli fw. 32.6.7 - flew more than 100 battery packs without this issue.
 Another thing is, from beginning there is problem that every 5 or 10 battery pack(wery random) motor 4 does not spin- esc initializing tone is fine for that motor, but arming the quad does not spin the motor - always motor 4, only power reset works.
video from motor 3 went full throttle after initializing: https://youtu.be/IU_nALgzGZM
setup: FC matek F722 STD with BF4 build 1544
esc holybro tekko 32 35A standalone
motors: hellgate furia 2700KV

sskaug, 2019-03-28:
Thank you for the detailed info, we'll look into it.
In the video I hear 3 beeps in the beginning - which are they? They are not the ESC powerup beeps or the signal detect beep or the armed beep.

ExaflopS, 2019-03-28:
First beep in the video was last beep from esc initialization tones and the 3 beeps  were buzzer beeps- acc calibration I think- So in order  last beep of esc initialization and buzzer beep together and another 2 buzzer beeps.

sskaug, 2019-03-31:
A test code Rev32.6.9 is now published. It now uses the number of good/bad frames over a given time as criteria for allowing to start the motor, instead of just timeout as was used in .8. This should provide a very reliable measure that signal is present and good before starting the motor.

ExaflopS, 2019-04-01:
@sskaug 
Thank you, I have tested new code Rev32.6.9 Now it sure is more reliable in terms of not spinning motor without arming, but problem is, If I try to arm and dirasm and arm again, motor 4, sometimes even motor 3and 4 wont arm, after disarming and waiting, the 2 esc restart and can be armed again- this could be issue during microfailsafe midair, taking away hope of flight recovery. I can do video demonstrating this issue if needed.

T3flon, 2019-04-01:
> @sskaug
> Thank you, I have tested new code Rev32.6.9 Now it sure is more reliable in terms of not spinning motor without arming, but problem is, If I try to arm and dirasm and arm again, motor 4, [...]

I have the same problem with the omninxt f7

sskaug, 2019-04-02:
Now a Rev32.6.1 test code is published (wrapping from 6.9 to 6.1)

ExaflopS, 2019-04-05:
@sskaug 
Thank you. I have tested new FW 32.6.1 and in terms of functionality it is similiar as 32.6.7 it just works good, but there is approximately 1/10 chance motor4 wont spin after arming. restarting FC does not help, esc for motor4 is somehow stuck and only power cyclyng by replugging battery helps- I dont know if this is BF issue or blehli issue or issue specific for MATEK F722 std fc

T3flon, 2019-04-05:
Motor 3 ist not running. Omninext f7

T3flon, 2019-04-09:
> 
> 
> Motor 3 ist not running. Omninext f7

FIXED

ExaflopS, 2019-04-23:
I have been testing 32.6.1 FW extensivley, I have flown few hundets of packs without issue of some motor spinning when it shouldnt. Thank You for fixing this issue.

closed 2019-04-23 by ExaflopS

eumelOne, 2019-04-26:
A Buddy had the same problem with just one motor spinnibg at fullthrottle without a arming signal. He has the 32.61 running. So for me it seems the problem is still existing.

sskaug, 2019-04-28:
Bidirectional dshot has become more complex than initially envisioned, involving switching between pullup and pulldown and neither. We have proposed to the BF test to change to an inverted dshot signaling for bidirectional dshot, and they have test code that runs very well in their initial testing. Hopefully BF can go for this, and we should have a much more reliable system.

yo8aiv, 2019-09-28:
hello.
 i had few similar cases on an old setup using dshot600, no testing version J-H-15 blheli_s
this **started to happen after i changed FC to an F7 JHEMCU** so this might be related. 

before this my quad was having following :
- kaute v1 running KAKUTEF4 (KTV1) 3.5.3 - replaced by  JHEF7DUAL (JH7D) 4.0.4 
- 30Aesc racerstar i think (has a current sensor)
- XSR rx
below esc conf. maibe it helps track down the issue.
`[ESC0]
Eep_Name=
Eep_ESC_Layout=J_H_15
Eep_ESC_MCU=#BLHELI$EFM8B21#
Eep_ESC_Mode=2
Eep_FW_Main_Revision=16
Eep_FW_Sub_Revision=7
Eep_Layout_Revision=33
Eep_Pgm_Startup_Pwr=9
Eep_Pgm_Direction=2
Eep_Enable_TX_Setup=1
Eep_Pgm_Comm_Timing=3
Eep_Pgm_Ppm_Min_Throttle=37
Eep_Pgm_Ppm_Max_Throttle=208
Eep_Pgm_Beep_Strength=40
Eep_Pgm_Beacon_Strength=80
Eep_Pgm_Beacon_Delay=4
Eep_Pgm_Demag_Comp=2
Eep_Pgm_Ppm_Center_Throttle=122
Eep_Pgm_Temp_Prot_Enable=7
Eep_Pgm_Enable_Power_Prot=1
Eep_Pgm_Brake_On_Stop=0
Eep_Pgm_LED_Control=0`