bitdump/BLHeli #354: Single motor run for a short time but then the esc´s restarts after arming.

T3flon, 2019-04-01:
After flashing and testing new beta fw blheli 32.6.9 on quad with omninxt f7 using firmware Airbot_Wraith32_ST_Multi_32_69  and flying with BF4 with RPM filters and bidir dshot enablet - using dshot 1200.
Single probs run for a short time but then the esc´s restarts after arming.

T3flon, 2019-04-01:
# diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Mar 26 2019 / 19:38:37 (173e958da) MSP API: 1.41

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id 

# name
name Alien 6s

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2075 0 0
aux 1 1 1 1475 2050 0 0
aux 2 13 3 1425 1575 0 0
aux 3 35 3 1725 1825 0 0
aux 4 36 2 1450 1550 0 0
aux 5 39 3 1950 2075 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 150
set gyro_to_use = SECOND
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -138,-142,-33
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 2000
set serialrx_provider = CRSF
set blackbox_p_ratio = 16
set dshot_idle_value = 450
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_switch_mode = STAGE2
set vbat_min_cell_voltage = 340
set ibata_scale = 188
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2081
set osd_tim_1_pos = 2103
set osd_current_pos = 2094
set osd_mah_drawn_pos = 2087
set osd_warnings_pos = 2441
set osd_esc_rpm_pos = 385
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_power = 3
set vtx_freq = 5658

# profile
profile 0

set dyn_lpf_dterm_max_hz = 540
set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 1

set roll_rc_rate = 201
set pitch_rc_rate = 201
set yaw_rc_rate = 168
set roll_expo = 33
set pitch_expo = 33
set yaw_expo = 10
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 64
set tpa_rate = 80
set tpa_breakpoint = 1750

RobMcFly, 2019-04-02:
exactly the same issue here

sskaug, 2019-04-02:
Now a Rev32.6.1 test code is published (wrapping from 6.9 to 6.1)

RobMcFly, 2019-04-02:
Thank you so much for the great work!!!!!!

RobMcFly, 2019-04-02:
Flashed and it worked. Testflight tomorrow

T3flon, 2019-04-05:
Rev32.6.1 dont work. Motor 3 is not running.

RobMcFly, 2019-04-05:
Flown 20 Packs without any issue

T3flon, 2019-04-05:
both quad have the problem. Engine 3 does not start ... 8/10. 1/10 engine 3 starts. 1/10 engine 3 only starts with throttle. Betaflight rc6. Quad 4s und 6s

T3flon, 2019-04-05:
without rpm filter everything is okey. https://www.youtube.com/watch?v=mHakjOKBP3c

T3flon, 2019-04-05:
> Flown 20 Packs without any issue

OMNINXT7?

RobMcFly, 2019-04-05:
Yes OMNINXT7 BF4.0 RC5 with RPM Filter: 
diff 

# diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Apr  1 2019 / 11:37:28 (88665a678) MSP API: 1.41

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id 

# name
name Rob FPV

# resources
resource MOTOR 1 B00
resource MOTOR 2 B05
resource MOTOR 3 B01
resource MOTOR 4 B04

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1475 0 0
aux 1 13 1 1300 1675 0 0
aux 2 35 1 1675 2100 0 0
aux 3 36 2 1475 2100 0 0
aux 4 49 3 1250 1650 0 0

# adjrange
adjrange 0 0 3 900 2100 29 3 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass2_hz = 150
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_derivative_hz = 150
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = SBUS
set dshot_idle_value = 800
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ESC
set vbat_scale = 111
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_warn_batt_not_full = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 1300
set osd_alt_alarm = 25
set osd_tim1 = 769
set osd_vbat_pos = 358
set osd_rssi_pos = 417
set osd_tim_1_pos = 14785
set osd_vtx_channel_pos = 8216
set osd_crosshairs_pos = 234
set osd_ah_sbar_pos = 234
set osd_ah_pos = 234
set osd_current_pos = 8277
set osd_mah_drawn_pos = 8311
set osd_craft_name_pos = 14337
set osd_display_name_pos = 11
set osd_altitude_pos = 471
set osd_warnings_pos = 14801
set osd_avg_cell_voltage_pos = 14817
set osd_esc_tmp_pos = 8345
set osd_esc_rpm_pos = 8408
set osd_stick_overlay_left_pos = 8451
set osd_stick_overlay_right_pos = 8465
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 6
set vtx_freq = 5843
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass_type = PT1
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 0
set smart_feedforward = ON
set iterm_relax_cutoff = 35
set p_pitch = 45
set i_pitch = 85
set d_pitch = 28
set f_pitch = 150
set p_roll = 48
set i_roll = 80
set d_roll = 30
set f_roll = 150
set p_yaw = 30
set i_yaw = 90
set f_yaw = 97
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 145
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 50
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end

#

T3flon, 2019-04-05:
resources
resource MOTOR 1 B00
resource MOTOR 2 B05
resource MOTOR 3 B01
resource MOTOR 4 B04

Maybe there is the key?

T3flon, 2019-04-05:
# diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Apr  3 2019 / 14:18:18 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id 

# name
name Mark 2 4s

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -LED_STRIP
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2100 0 0
aux 1 1 1 1475 2050 0 0
aux 2 13 3 1425 1575 0 0
aux 3 35 3 1725 1825 0 0
aux 4 36 2 1450 1550 0 0
aux 5 39 3 1950 2075 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 150
set gyro_to_use = SECOND
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -139,-138,-37
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 2000
set serialrx_provider = CRSF
set blackbox_p_ratio = 16
set dshot_idle_value = 450
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_switch_mode = STAGE2
set ibata_scale = 188
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2081
set osd_tim_1_pos = 2103
set osd_current_pos = 2094
set osd_mah_drawn_pos = 2087
set osd_warnings_pos = 2441
set osd_esc_rpm_pos = 385
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 3
set vtx_freq = 5880

# profile
profile 0

set dyn_lpf_dterm_max_hz = 540
set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 1

set roll_rc_rate = 201
set pitch_rc_rate = 201
set yaw_rc_rate = 168
set roll_expo = 33
set pitch_expo = 33
set yaw_expo = 10
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 64
set tpa_rate = 80
set tpa_breakpoint = 1750

RobMcFly, 2019-04-05:
i

> resources
> resource MOTOR 1 B00
> resource MOTOR 2 B05
> resource MOTOR 3 B01
> resource MOTOR 4 B04
> 
> Maybe there is the key?

i don´t think. I only turned the board in this build.

T3flon, 2019-04-05:
So esc's flashed again, with the same negative result. engine 3 did not start.
With betaflight 4.0 RC6 + 32.6 the quad runs as usual

T3flon, 2019-04-09:
FIXED. Android App update!

closed 2019-04-09 by T3flon