bitdump/BLHeli #356: Seversl bits are missing in telemetry block

alxsfrm, 2019-04-08:
Hi,
I'm working with 32.6 version on my own driver. I flashed the firmware about two or three months ago as 'latest'.
Sending Dshot command at 500Hz, and Telemetry request flag set at 100Hz.
When i'm trying to read the Telemetry data, i get only 9 bytes instead of 10.
On oscilloscope i see that the packet comes with several bits of last byte missing.
Is this problem known to someone ?

sskaug, 2019-04-09:
Have you tried faster Dshot signal rate? Dshot is not designed to run below 1kHz (for a 48MHz MCU) due to a 16bit counter being used.

alxsfrm, 2019-04-09:
Hi sskaug,
I just followed the manual, which says that "For a 48MHz clock MCU, the minimum input signal rates are about 40Hz for 1-2ms PWM, about 
1000Hz for Proshot and about 300Hz for all other input signal types.".
I will try your recommendation.

sskaug, 2019-04-09:
Ah... sorry about that. I have had it on my list for some time to update the manual - will have to do that soon.

robert-hoffmann, 2019-04-14:
Sorry if this question sounds stupid...

> Dshot is not designed to run below 1kHz (for a 48MHz MCU) due to a 16bit counter being used

Does this mean that running dhsot600 on a F3 furling ESC is not ideal ?

alxsfrm, 2019-04-16:
> Have you tried faster Dshot signal rate? Dshot is not designed to run below 1kHz (for a 48MHz MCU) due to a 16bit counter being used.

I tried to send the commands at about 5 kHz, and i get following results:
According to the oscilloscope, i get telemetry blocks at 100Hz (as expected), but the size of the blocks changes between 4.2 to 5.2 ms, which corresponds to 48 to 60 bytes of data. If i zoom in, i really see bit size of 8.6 uS (115200 bps).
As far as i know, each telemetry block has size of 10 bytes.
So there are two questions: What is the structure of the big block ? and how it is possible that the blocks has different sizes ?

sskaug, 2019-04-16:
The length of the response is about 860us. I do not know what happens in your case, when I hook it up to Betaflight (or KISS FC) it runs as it shall. Maybe you can take a look at those setups?

alxsfrm, 2019-04-17:
I can take a look at Betaflight implementation, but you sure that under any circumstances you're not joining telemetry blocks of same request, or put an additional data on?

alxsfrm, 2019-05-06:
Sorry, it was my bug in buffer management.
It's working now.
Thank you.

closed 2019-05-06 by alxsfrm