bitdump/BLHeli #358: Arduino BL Bootloader Vs 4Way Interface

skorzo, 2019-04-14:
Hi there, this is is my problem.

I have 4 x Furling32 Blheli32 ESCs  running Blheli32 32.4  (Bootloader h)

The 4 ESCs are running fine, no problem with my quadcopter.

The problem is that I tried to update the firmware and I cant connect to ESC 3.

(see the picture, no mistery --> https://i.ibb.co/frKdKSk/Sin-t-tulo-1.jpg )

I have tried:
-   Betaflight Interface  (Omnibus F4 V6. Quad is running fine, but no ESC 3 on blheli32 suite)
- Arduino Nano 4way interface   (I have tried the 3 files that you can flash) -> Same result
- Arduino Blheli Bootloader  -> Same result, no ESC 3

Any suggestion?

By the way... which is the difference between trying to connect via Arduino 4way interface and "Arduino Blheli bootloader" ?  I cant find information.

Lots of times I have managed to flash No-Blheli32 Escs with my 4way interface but Blheli32 seems to be different.

Any Ideas? Thanks

4712, 2019-04-14:
@skorzo:  Try to "Connect" first, and power ESC afterwards.
BTW: Arduino Blheli bootloader emulates an USB/UART bridge for single ESC connection, usage with Interface type "L".
Maybe have a look at: https://github.com/bitdump/BLHeli/blob/master/Atmel/BLHeli%20programming%20adapters.pdf

skorzo, 2019-04-15:
> @skorzo: Try to "Connect" first, and power ESC afterwards.
> BTW: Arduino Blheli bootloader emulates an USB/UART bridge for single ESC connection, usage with Interface type "L".
> Maybe have a look at: https://github.com/bitdump/BLHeli/blob/master/Atmel/BLHeli%20programming%20adapters.pdf

Thanks for the link, altought I dont see why to choose 4way or Interface type L when trying to flash my esc with an Arduino.

I tried to "Connect" first and power ESC afterwards, but ESC is still not found.  ( In fact that is the way I always tried, so I tried to power ESC and then connect, but same result)

How is it possible that my ESC is "bricked" and  I cant flash it but anyway it is running OK when flying the quadcopter?