bitdump/BLHeli #365: test code 32.61 causes weird issue when flashing

derekkeithchan, 2019-04-28:
I flashed ESC 1 with the test FW 32.61 and it says its not compatible with the blheli suite 32 firmware. i am using the latest firmware (32.6.1.2). I am also unable to revert it as when i click flash it says firmware no compatible again. refer to picture.
[image]

4712, 2019-04-28:
@derekkeithchan Did you try to cycle power of the ESC and "Check" again?

derekkeithchan, 2019-04-28:
> @derekkeithchan Did you try to cycle power of the ESC and "Check" again?

yes, multiple times :/

4712, 2019-04-28:
@derekkeithchan Please check the attached beta exe. 
[BLHeliSuite32.zip](https://github.com/bitdump/BLHeli/files/3124946/BLHeliSuite32.zip)

derekkeithchan, 2019-04-28:
@4712 Thanks for that, this time it read as unknown firmware and allowed me to erase and reflash. however, the test firmware still didnt work.

derekkeithchan, 2019-04-28:
@4712 forgive me.. i just tried flashing with this beta exe and it flash fine. thanks!

closed 2019-04-28 by derekkeithchan

4712, 2019-04-28:
The issue was probably caused by a wrong saved parameter layout version byte value.
> 
> 
> @4712 Thanks for that, this time it read as unknown firmware and allowed me to erase and reflash. however, the test firmware still didnt work.

Weird, it should have worked after erase and power cycling even with the old 32.6.1.2 version...
> 
> 
> @4712 forgive me.. i just tried flashing with this beta exe and it flash fine. thanks!

Nothing to forgive, thank you for reporting the issue! 
It helps to ongoing improve the error checking procedures.