bitdump/BLHeli #366: BLHeliSuite32xl crashes

andrejpodzimek, 2019-04-28:
```
$ ./BLHeliSuite32xl 
Exception EAccessViolation in module BLHeliSuite32xl at 000000000043D65D.
Access violation at address 000000000043D65D, accessing address 0000000000000000.
```

Sometimes (~1 attempt out of 5) it just gets a segmentation fault wihtout even trying to fail gracefully.

The version I tried is `BLHeliSuite32xLinux64_Beta_0904.zip`. OS: ArchLinux, stock kernel 5.0.7. (The only unusual thing about my machine (just in case it matters) is that it has two 5K monitors. Not sure this can be related to graphics in any way.)

Also, BTW, BLHeli32 for Windows runs OK on the same machine under Wine. But Wine doesn't have a good high-DPI compensation nowadays, so you need a magnifying glass to use it on a 5K monitor.

4712, 2019-04-28:
@andrejpodzimek Does it crash at startup, or when a specific action is executed?

andrejpodzimek, 2019-04-28:
@4712 Immediately at startup; before anything X11-related appears.

4712, 2019-04-29:
It runs fine here with arch linux vbox image, may be it's related to the screen resolution...

andrejpodzimek, 2019-04-29:
That might be the case if something overflows and (e.g.) accesses a negative array offset or the like. This is what `gdb` says when it comes to the crash. It's closed-source :stuck_out_tongue_closed_eyes:, but maybe you can match it against your debugging symbols somehow:

```
Thread 1 "BLHeliSuite32xl" received signal SIGSEGV, Segmentation fault.
0x0000000000837248 in ?? ()
(gdb) bt
#0  0x0000000000837248 in  ()
#1  0x0000000000836562 in  ()
#2  0x0000000000837f16 in  ()
#3  0x0000000000837cbe in  ()
#4  0x0000000000837b98 in  ()
#5  0x0000000000837a27 in  ()
#6  0x0000000000837db2 in  ()
#7  0x000000000087f23c in  ()
#8  0x0000000000880480 in  ()
#9  0x00000000009d5e70 in  ()
#10 0x0000000000a035d7 in  ()
#11 0x00000000007ee2a8 in  ()
#12 0x00000000007ee189 in  ()
#13 0x00000000007887eb in  ()
#14 0x00000000007eeb95 in  ()
#15 0x0000000000a0ad4d in  ()
#16 0x0000000000000001 in  ()
#17 0x00007fffffffe83b in  ()
#18 0x0000000000000000 in  ()
```
[This is a `strace -f` output from a launch attempt.](https://github.com/bitdump/BLHeli/files/3125686/blheli_strace.txt) It might give you more information on where exactly it crashed. When I launch it with `strace`, it segfaults all the time. When I launch it directly, without `gdb` or `strace`, it is highly likely to end with the exception above and segfaults occur only in one attempt out of many.

4712, 2019-04-29:
@andrejpodzimek  Thanks, it looks like something fails with usb port scan.
Some executables to test:
1: HID disabled
2: HID and serial Port scan disabled
3: HID and serial Port disabled (no connect possible)
4: HID and serial Port disabled (no connect possible) and initial Form resize disabled.
 
[Test1NoHID.zip](https://github.com/bitdump/BLHeli/files/3127508/Test1NoHID.zip)
[Test2NoPortTimer.zip](https://github.com/bitdump/BLHeli/files/3127516/Test2NoPortTimer.zip)
[Test3NoPortAccess.zip](https://github.com/bitdump/BLHeli/files/3127519/Test3NoPortAccess.zip)
[Test4NoFormSize.zip](https://github.com/bitdump/BLHeli/files/3127522/Test4NoFormSize.zip)

andrejpodzimek, 2019-04-30:
Thanks for looking into this. First and foremost, these tests don't crash; they all keep running.

### The first test

This freezes forever and never redraws the window content. The window remains black or just contains whatever had been under it when it started. [Here's an `strace -f` of that.](https://github.com/bitdump/BLHeli/files/3129750/strace.Test1NoHID.txt) Because I didn't want to send too many events to the window (which may bloat the `strace` output), I started it like this and then let it sit until it got killed:

```bash
timelimit -t 120 -T 180 strace -f ./BLHeliSuite32xl 2> strace.Test1NoHID.txt
```

### The second test

This appears to be working. It connected to an FC just fine. However, it takes an extremely long time to start. Once it starts, things look OK. As the test's name seems to suggest, it doesn't update the list of ports dynamically, so the FC has to be plugged in before the app runs.

This is with no FC plugged in:
[image]

This is with an ancient F3 plugged in and connected. Everything looks normal:
[image]

### The third test

It starts instantaneously and shows the full interface, just like above, and everything looks OK. (Apart from the fact that one can't select ports, so indeed, as you say, no connection there.) But it starts (AFAICT) within a second, unlike the (working) second test that takes tens of seconds to start.

### The fourth test

Surprisingly enough, this just freezes and doesn't redraw the window. It behaves in a way similar to the first test, AFAICT. [Here's an `strace -f` of that](https://github.com/bitdump/BLHeli/files/3129779/strace.Test4NoFormSize.txt), taken exactly the same way as in the first test.

4712, 2019-04-30:
@andrejpodzimek Thank you very much, I will need some time to look into the results and come back with a hopefully fixed version.
Perhaps the long start time  of Test2 is caused by initial checking for updated version, which can be disabled under "Options". Or again because of the serial port scanning, which tries to open each port to get a list of  "real existing" ports.

andrejpodzimek, 2019-04-30:
Ports on my machine might be actually surprising to a contemporary app, because it's a Supermicro server motherboard ([this one in particular](https://www.supermicro.com/products/motherboard/Xeon/D/X10SDV-16C_-TLN4F.cfm)) that has one of those old-fashioned "real" serial ports. I'm not using it for anything, but it's there. (The GPU is an NVidia Quadro P5000 with the NVidia binary driver, just in case it matters to the widget toolkit you are using.)

djabi, 2019-12-17:
I think I'm getting the same crash. Its 100% repro. It happens when I select "Select BLHeli_32 Interface > BLHeli32 Bootloader (4way if)". The last few lines in the strace -f before the crash are similar to the trace posted by andrejpodzimek:

[pid  4351] lstat("/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.2/bInterfaceNumber", {st_mode=S_IFREG|0444, st_size=4096, ...}) = 0
[pid  4351] openat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.4/2-1.4:1.2/bInterfaceNumber", O_RDONLY|O_CLOEXEC) = 35
[pid  4351] fstat(35, {st_mode=S_IFREG|0444, st_size=4096, ...}) = 0
[pid  4351] fstat(35, {st_mode=S_IFREG|0444, st_size=4096, ...}) = 0
[pid  4351] read(35, "02\n", 4096)      = 3
[pid  4351] read(35, "", 4096)          = 0
[pid  4351] close(35)                   = 0
[pid  4351] --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---

4712, 2019-12-18:
@djabi Could you try to edit the BLHeliSuite32xl.ini file in the /settings folder and try what happens after next start?
BTW please use version 0.9.1.7...
> [Interface]
AutoPortScan=0
TestPortOnEnum=0

djabi, 2019-12-21:
Yep, adding the lines in the ini file makes it not crash. I'm using 0.9.1.7 on Ubuntu 19.10 with 5.3.0-24 in case that matters.

4712, 2019-12-21:
Thanks a lot for your feedback.
I guess "TestPortOnEnum=0" does the job.
The Ubuntu system revision is not important, I think.
It's more about the hardware/driver configuration of the machine which includes a serial port, which lets the the app crash, when it tries to test the port (open the port).
Maybe you can post the file "/proc/tty/drivers" here?

4712, 2019-12-21:
Got an idea to catch the exception. 
Please try the attached beta:
[BLHeliSuite32xLinux64_Beta_09171.zip](https://github.com/bitdump/BLHeli/files/3991719/BLHeliSuite32xLinux64_Beta_09171.zip)
with
AutoPortScan=1
TestPortOnEnum=1

Thanks