bitdump/BLHeli #367: usb_hid_cdc doesn't work on MatekF411RX BF4.0

ljlzonda, 2019-05-08:
I've got a MatekF411RX board, with built in SPI receiver for Frsky/Futaba. I run SFHSS mode to bind to my Futaba radio. I've tried 4.0.0 and 4.0.2, neither will the usb_hid_cdc work with liftoff. Liftoff can detect the controller but can't read out the data of each channel.

mikeller, 2019-05-08:
@ljlzonda: You are not going to get help for this issue here. Can you please close this, and reopen on https://github.com/betaflight/betaflight/issues?

ljlzonda, 2019-05-08:
sorry mate. I messed up.发自我的iPhone------------------ Original ------------------From: Michael Keller  <notifications@github.com>Date: Thu,May 9,2019 7:38 AMTo: bitdump/BLHeli <BLHeli@noreply.github.com>Cc: ljlzonda <1500011309@pku.edu.cn>, Mention <mention@noreply.github.com>Subject: Re: [bitdump/BLHeli] usb_hid_cdc doesn't work on MatekF411RX BF4.0(#367)@ljlzonda: You are not going to get help for this issue here. Can you please close this, and reopen on https://github.com/betaflight/betaflight/issues?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/bitdump/BLHeli/issues/367#issuecomment-490689740",
"url": "https://github.com/bitdump/BLHeli/issues/367#issuecomment-490689740",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2019-05-10 by ljlzonda