bitdump/BLHeli #370: Simple ESC

Andybotjustice, 2019-05-16:
Has anyone found or made a simple version of BLheli that can be compiled to just run a brushless motor forward/reverse?  I was thinking something like a DSHOT setup that takes no calibration and simply will run forward for a range of inputs and reverse for a range of inputs.   No protection / telemetry / etc... turn the BLDC into something controlled more like a normal brushed motor.  Needed for other BLDC projects that calibration / special features cause an issue for...   

I posted the task on freelancer but really didn't see anyone interested and capable of doing it...

danyaPostfactum, 2019-10-24:
It's quite easy. Just remove initial beeps, arming check and calibration procedure from the code. I need almost the same thing, but I'd like to control a motor by a potentiometer (one direction).

bartwaw, 2022-07-18:
Someone managed to modify the firmware in this way ?