bitdump/BLHeli #371: Sine modulation mode issue BLHeli32 V32.61 and 32.62

Gregmaan, 2019-05-22:
With the newer beta versions the sine modulation modus is not useable anymore:
Grinding motors, and with Betaflight 4.02 bidir on, some motors do not start on arming.
BLHeli32 runs mostly on defaults exept timing:auto and sine modulation mode: on

sskaug, 2019-05-23:
I've tried to reproduce this before and now once again. With no luck.
I run on the bench and it runs fine, I can not hear any grinding.
What dshot speed and signal rate do you use?

Gregmaan, 2019-05-23:
Hmm strange. I used DShot 600 @ 4k. Thats the recommendation from the Betaflight guys for the bidir telemetry. On first arm allmost every motor starts on arming. But after disarm and rearm the problem gets worse (1 to 3 motors did not start). As soon as I disable sine modulation all is fine.
Tested with 2 quads Matek F405 board with Aikon 35A and Kakute F7 board with Aikon 30A.
Maybe its related to the auto timing?

sskaug, 2019-05-27:
Ah! Now I reproduce it. Obviously it depends on the timer configuration of the ESC, so when testing with the proper ESC, motor grinds. I'll dig into it.

sskaug, 2019-06-03:
Now new test code 32.6.4 and 32.6.5 are published, where this is fixed.

Gregmaan, 2019-06-06:
Seems fine for me in both cases. Thanks!

closed 2019-06-06 by Gregmaan