bitdump/BLHeli #373: Pyro32 f3 rpm invalid packet on motor 2 using inverted fw

dronejunkie, 2019-05-27:
Fc is aikonf4 30x30
Betaflight fw nightly build number 1663
Motor 2 consistently has more than 10% invalid packets. 

Dshot reads: 436640
Dshot invalid pkts: 37037
Dshot irq micros: 3

Motor     RPM   Invalid
=====   =====   =======
    0       0     0.00%
    1    2047    13.16%
    2       0     0.03%
    3       0     0.00%

7 7 7 7 7 7 7 7 7 7 7 7 14 14 14 14 0 0 13 1 15 3 17 5 0 15 2 17 4 19 6 2
0 0 0 0 0 0 0 0 0 8 8 8 15 15 15 16

The uninverted version bidirectional dshot has no invalid packet. 

MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.32V, Core temp=49degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3024, Max available config: 16384
GYRO=ICM20602
System Uptime: 133 seconds, Current Time: 2019-05-27T11:49:04.381+00:00
CPU:11%, cycle time: 250, GYRO rate: 4000, RX rate: 33, System rate: 10
Voltage: 1533 * 0.01V (4S battery - OK)
I2C Errors: 0
Arming disable flags: RXLOSS NOPREARM CLI MSP
Dshot reads: 2019530
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

faceforfpv, 2019-05-28:
Similar results on a Furling32 4in1. FC is Matek722.
In this case, two motors have an invalid packet rate of more than 10%, consistently.

# dshot_telemetry_info
Dshot reads: 7813644
Dshot invalid pkts: 5669620
Dshot irq micros: 3

Motor     RPM   Invalid
=====   =====   =======
   0    2047    16.55%
   1       0    17.33%
   2       0     0.00%
   3       0     0.00%

10 8 16 11 18 13 0 15 3 17 5 19 7 2 9 4 12 6 14 8 16 11 18 13 1 15 11 17 13 19 15 2
18 15 15 15 14 14 15 15 14 14 15 15 14 6 6 7

dronejunkie, 2019-05-28:
This issue appears on a number of f3 esc.  Tekko f3
https://github.com/bitdump/BLHeli/issues/374

dronejunkie, 2019-06-05:
On the bench there is no error but during flight there are clusters of errors.  The pattern is during punch out.   Here is the BBL https://drive.google.com/open?id=1L-QLArWohQiqjMlI1EvbkF90NxbjoTwo 
I set debug to  dshot rpm error.   I am using dshot600,  I have not try dshot1200.