bitdump/BLHeli #374: lost packets for inverted dshot with rpm filters on blheli 32.63 for tekko F3 metal

ExaflopS, 2019-05-27:
I found issue with using latest fw with inverted dshot 32.63 TEKKO32_F3_Metal_Multi_32_63.Hex with matek F722 STD on BF build 1664. 
There is problem with some motors not reporting correct telemetry. This problem is not present when I use earlier uninverted blheli fw like 32.67 

[image]
[image]

sskaug, 2019-06-03:
Now new test code 32.6.4 and 32.6.5 are published, where this is fixed.

ExaflopS, 2019-06-04:
@sskaug  Thank you, I have tested blheli FW 32.6.5, the described issue is fixed.

closed 2019-06-04 by ExaflopS